MJoergen/C64MEGA65 #1: Compatibility issues of unknown origin

sy2002, 2022-04-08:
The following software is known to run on MiSTer but not on our core.

The MiSTer version used for testing was:

    C64_20210701
    Plain kernal/rom/1541 (from C64 Forever)

This is exactly the MiSTer version we use, so therefore this is an odd thing:

* appf1: appf1.d64
* PolarBear: PolarBearInSpace_11HDG-[EX].zip

While there are also compatibility issues related to our 99.75% system slowdown (see issue xyz) the software listed here was tested using the original clock speed and still does not work on our core

sy2002, 2022-05-13:
Note to self: Might be related to the fact, that we currently have a read-only core and also set the "read-only" flag for the C1541.

lydon42, 2022-05-17:
Commando (any release): joystick button does not work

glbyers, 2022-06-07:
FYI, another one that won't run on this core; arkanoid [1351]

https://csdb.dk/release/?id=162087

sy2002, 2022-06-07:
@glbyers Thank you for reporting. Can you please re-test but switch off "HDMI: Flicker-free" before you load the game? (I am not near a MEGA65 for the next 3 weeks, otherwise I would have done this by myself ;-) )

glbyers, 2022-06-07:
OK, no problem. Tested with HDMI: Flicker-free disabled, but same result unfortunately :(

sy2002, 2022-06-07:
@glbyers One more question: I just looked at the CSDB.DK release: "won't run" means that the game does not start or that the mouse does not work? At it's current state, the C64 core does not support the mouse, yet.

glbyers, 2022-06-07:
It won't actually start. Although if there is no mouse support yet, that may be causing the failure?

sy2002, 2022-06-08:
I am not sure - we will need to investigate this further. Thank you for taking the time and reporting the issue.

sy2002, 2022-06-18:
Demo: "Desert Dream" (https://csdb.dk/release/?id=48039):
* Runs flawlessly
* But does not detect disk change (you need to manually load the parts)
* Double-checked if flicker-free is the reason: No, it is not
* Tested with "Release 1" of the C64 core and a release candidate of "Release 2"

sy2002, 2022-06-19:
Regarding the above-mentioned (https://github.com/MJoergen/C64MEGA65/issues/1#issuecomment-1148397299) Arkanoid issue from glbyers, Discord user JudgeDrokk shared the following info including the ZIP with me:
[ArkanoidVersions.zip](https://github.com/MJoergen/C64MEGA65/files/8936254/ArkanoidVersions.zip)
```
Hi - in relation to the Arkanoid issue as mentioned in a convosation in "other-cores" , Deft suggested that you were the right person to send these to.  These are the four versions of Arkanoid that I tried and I've marked the one that worked.   The others would not execute the game after unpacking, or after the crack intro screen, I'm sure that normally the game boots to the title screen and classic Martin Galway tune, but that is nowhere to be seen in the WORKING version, it jumps straight into the control selection before and after each game.    Thanks.

Arkanoid+-TUC(WORK).d64 is the only version I have been able to run. I have now tried 6 versions, including a 1351 fix version someone linked me from CSDB.   The latest version I tried was by REMEMBER +3 trainer.   Intro and trainer appeared correctly but the game would not start after the trainer option screen.
```

sy2002, 2022-06-20:
Discord user Maurice#4307 confirmed on version 2 and Discord user JudgeDrokk#5308 confirmed on version "resetfix-100mhz" that Bomberman (this version: https://csdb.dk/release/?id=45809) has these effects (quote):
```
It boot and start without issues, but it seems an invisible source is pressing the fire button continuously.....
Even throughout the menu......
```

sy2002, 2022-06-21:
Release Candidate 2 of Version 3 fixes:

* Demo Apparatus works now: appf1.d64 (see https://github.com/MJoergen/C64MEGA65/issues/1#issue-1197100781)
* Demo Comalight works now
* Commando (see https://github.com/MJoergen/C64MEGA65/issues/1#issuecomment-1129271045): Fire button works now

I expect a lot of more games/demo to work now; this one has a huge impact.

The RC2 is here:
https://github.com/MJoergen/C64MEGA65/tree/673adcb648bb290b31e86efad18213fb27dabd50/bin/Version%202/R3

sy2002, 2022-06-22:
**Arkanoid** is fixed with Release Candidate 3 of Version 3:
https://github.com/MJoergen/C64MEGA65/blob/fb1172f4a5f90e07de9f7e9ff7a6e740ce0f53d6/bin/Version%202/R3/README.md

sy2002, 2022-06-22:
All issues are resolved. We will use separate trackers per game/demo from here on.

closed 2022-06-22 by sy2002