MJoergen/C64MEGA65 #3: en-/disable sound via digital video output for DVI

cwscws, 2022-04-26:
The Mega65 itself provides an option in its BIOS to enable or disable the sound for DVI. The C64 core doesn‘t. Using a HDMI-to-DVI cable. Usually I get no picture on my display if the output is not set to DVI.

sy2002, 2022-04-26:
Thank you for letting us know. This is a rather easy (not complicated) thing to do, so we will happily consider this for our next release.

sy2002, 2022-05-13:
Reminder to self: For this and other DVI/HDMI compatibility things (such as force 60 Hz), we need some config file (".ini" like MiSTer?) where people can configure this, because otherwise they would need to do it "blindly" pressing Help and then other key combos without seeing anything on screen.

sy2002, 2022-05-23:
We are preparing a bugfix and stability release for the C64 core. I wonder, if it would make sense to add the DVI topic there.
@cwscws Are you able and willing to test an experimental core, which is "DVI only" (i.e. I am about to create an experimental version of the core that disables the "soundtrack" inside the data stream and sets the DVI flags properly).

sy2002, 2022-05-23:
[dvitest.zip](https://github.com/MJoergen/C64MEGA65/files/8754408/dvitest.zip)

Here is a test `.cor`and `.bit`core that (should) do proper DVI (and only DVI, no sound in the HDMI data stream).

sy2002, 2022-06-18:
This feature request was implemented in version 2:
https://github.com/MJoergen/C64MEGA65/blob/V2/VERSIONS.md

closed 2022-06-18 by sy2002