MJoergen/C64MEGA65 #4: No video is displayed on a Philips 22PFT5303_05/5300 TV/monitor

grim-fandango, 2022-05-10:
No video is displayed in the C64 core on the above monitor - the monitor reports "No signal" on all permutations of the 60Hz and flicker-free toggles in the core menu.

In contrast, MEGA65 mode displays fine.  The monitor has the latest firmware installed.

Monitor page is here: https://www.philips.co.uk/c-p/22PFT5303_05/5300-series-full-hd-ultra-slim-led-tv

sy2002, 2022-05-10:
@gardners All the results of our thorough HDMI debugging and fine-tuning session that we had back in 2021 together with Adam Barnes from the Tyto Project (https://github.com/amb5l/tyto_project/issues?q=is%3Aissue+is%3Aclosed) have been added to the MiSTer2MEGA65 framework so I would assume that the C64 MEGA65 continues to pass the tests of your HDMI analyzer.

But: We might have introduced bugs when we started to add MiSTer's "ascal.vhd" to our Audio- / Video pipeline.

This case here triggered me: I want to be sure, that the core *still* passes all the tests of your HDMI analyzer. Could you run a test again, please with this bitstream:

https://github.com/MJoergen/C64MEGA65/blob/master/bin/Version%201/R3/C64MEGA65-1-R3.bit

MJoergen, 2022-05-10:
I found the user guide for the monitor mentioned here: (https://www.download.p4c.philips.com/files/2/22pft5303_05/22pft5303_05_dfu_eng.pdf).
In section 13.5 "Display Input Resolution" the following is stated:

```
For PFx series TV
Video formats
Resolution — Refresh rate
- 480i - 60 Hz
- 480p - 60 Hz
- 576i - 50 Hz
- 576p - 50 Hz
- 720p - 50 Hz, 60 Hz
- 1080i - 50 Hz, 60 Hz
- 1080p - 24 Hz, 25 Hz, 30 Hz
Computer formats
Resolutions (amongst others)
- 640 x 480p - 60 Hz
- 800 x 600p - 60 Hz
- 1024 x 768p - 60 Hz
- 1280 x 768p - 60 Hz
- 1360 x 768p - 60 Hz
- 1280 x 1024p - 60 Hz
- 1920 x 1080p - 60 Hz
```
I'm confused about why the user guide differentiates between "Video formats" and "Computer formats", but in the former section the entry "720p" appears, so from that I conclude that the monitor does indeed support the output format supplied by the core.

grim-fandango, 2022-05-10:
There is an option in the advanced swings for turning computer mode off and on. Presumably it chooses between the two sets of supported modes you list above.  However even with it switched off the C64 core doesn’t display.  

In any case since this is a corner case we can close this issue I guess?

[image]

sy2002, 2022-05-10:
@MJoergen and @grim-fandango thank you for your feedback.

Just for being on the safe side (and as stated in Discord, we are not in a hurry here): Let's keep this issue open and as soon as @gardners has time in a few weeks or months to redo the test with the HDMI analyzer I propose to proceed like this:

* If @gardners tests are showing that the core behaves "OK" then I would indeed close this one, because the probability that we find anything "blindly" without access to an industry-grade new HDMI analyzer is pretty low.

* But if @gardners tests are showing some anomaly, then we can work on it an @grim-fandango then it would be great, if you would still own this monitor and could support us with your tests.

grim-fandango, 2022-05-12:
Not sure if this helps or not, but the same monitor's VGA input works fine.  The general release MEGA65 (i.e. not the devkit, the one in the proper case) doesn't work in HDMI mode either (not that I was expecting it to!).  Changing the HDMI cable didn't help either.

sy2002, 2022-05-13:
Thank you @grim-fandango ! Can you do one more test with this monitor please, using the MEGA65 core? It would be: Switch the MEGA65 core to DVI mode. If this works, then this issue here would be a duplicate of issue #3

grim-fandango, 2022-05-13:
I changed it from enhanced DVI to just DVI - I assume that's what you meant,  I get a MEGA65 mode display in both enchanced and audio only modes.  This is through an HDMI cable though, and not an HDMI-to-DVI cable as in #3

sy2002, 2022-05-15:
Just for clarity @grim-fandango: Talking about the MEGA65 core (not the C64 core), did I understand you correctly, that on this very monitor, the HDMI mode is not working (see https://github.com/MJoergen/C64MEGA65/issues/4#issuecomment-1125509708) and that as soon as you switch to DVI, the MEGA65 core is working?

grim-fandango, 2022-05-16:
The monitor (the Philips one, which doesn't show video on the C64 core) only has HDMI and VGA ports, not DVI.  When you're talking about DVI mode, do you mean this?

[image]

sy2002, 2022-05-16:
Yes, I do.

What I did not fully get was this statement from you:

"The general release MEGA65 (i.e. not the devkit, the one in the proper case) doesn't work in HDMI mode either (not that I was expecting it to!). Changing the HDMI cable didn't help either."

So you did not get any visible HDMI output?

And when you switched now to "DVI only (no audio)" using a HDMI cable: Did the MEGA65 core (not the C64) show any picture? Because you wrote

"I changed it from enhanced DVI to just DVI - I assume that's what you meant, I get a MEGA65 mode display in both enchanced and audio only modes."

And this confuses me, because it contradicts the other statement when you said "doesn't work in HDMI mode either".

grim-fandango, 2022-05-16:
Talking about two cores is confusing :-)

The monitor works fine on the Mega65 core for both DVI modes.

The monitor doesn’t work at all on the C64 core. I tried a different HDMI cable, and tried both the white case M65 and the devkit.

The monitor *does* work in the C64 core using VGA.

Hope that makes more sense :-)

Jason

sy2002, 2022-05-16:
Hi Jason, Thank you - now I got it ;-)
That means:
1. This issue here is *not* a duplicate of issue #3 
2. Let's proceed as described in comment https://github.com/MJoergen/C64MEGA65/issues/4#issuecomment-1122888369

iFlash, 2022-06-08:
Hi there,

just adding another TV that refuses to show a picture for the c64 core: The Samsung "The Frame" 2021 edition. I have tried the DVI-core, too, to no avail. The picture does show on my small Blackmagic monitor, the CRT emulation switch in the Help menu has no effect, though (not sure if that has not been implemented yet).

Hope that helps,

Armin

sy2002, 2022-06-09:
@iFlash Thank you for sharing. I assume that you connected the Blackmagic via VGA? Because CRT emulation actually is implemented only for HDMI (and looks great on most monitors)

sy2002, 2022-06-09:
About The Frame: It is a high end 4K TV - not sure if it accepts anything lower than full HD? (We output 720p only)
I also heard people being successful while trying older HDMI cables (strangely enough).

Anyway: we will report back as soon as we did some more tests with the HDMI analyzer.

sy2002, 2022-06-09:
Aha! We are doing negative/negative and this might be it?
We need to double-check if what we are doing is confirming to the standard or not.

@MJoergen interessting!

@iFlash I will produce a positive/positive polarity core for you to test

sy2002, 2022-06-09:
@iFlash and @grim-fandango This is just a shoot in the dark, playing around with different "polarity settings"; please try this core on your non-working HDMI monitors/TVs in 50 Hz mode as well as in 60 Hz mode

@MJoergen What I wrote above is not accurate, we seem to do everything correctly, but hey, let's experiment by just reversing the polarity. Maybe it helps some people
[hdmi-exp1.zip](https://github.com/MJoergen/C64MEGA65/files/8874438/hdmi-exp1.zip)
.

iFlash, 2022-06-10:
@sy2002 Here are some new findings:

My small Blackmagic Monitor will only show a picture booting to c64 cores (incl. the hdmi-exp1 from above) but will stay black with mega65 cores. "The Frame" will only provide a picture with MEGA65 cores but not with any 64 core including the hdmi-exp1.

grim-fandango, 2022-06-10:
The core hdmi-exp1.zip doesn't work on either my good monitor or the DVI monitor.  It gives a blank screen/ no signal on both.

sy2002, 2022-06-10:
@iFlash and @grim-fandango Thank you for your time and for testing the experimental core. Admittedly we are totally groping in the dark when it comes to these phenomenon... Nevertheless: It was worth a try :-)

iFlash, 2022-06-10:
@sy2002 As the MEGA65 core shows no issues, isn't it possible to copy what they do regarding HDMI output?

I have no insight if that is even a suitable approach, so forgive me if this is a stupid question.

sy2002, 2022-06-17:
@iFlash No, the MEGA65 core uses a different resolution: The C64 core is 720p on HDMI (1280×720) while the MEGA65 is PAL or NTSC via HDMI (720x576 or 720x480). This difference might not sound like a big deal, but it is.
Learn more about the C64 core's video output here: https://github.com/MJoergen/C64MEGA65#video-and-audio

ChristianWeyer, 2022-07-02:
I have a similar problem with my Samsung UE32J5150 TV (https://www.samsung.com/de/support/model/UE32J5150ASXZG/).
Should I open a new issue, or keep it here @sy2002 ?

Thanks!

sy2002, 2022-07-02:
@ChristianWeyer Thank you for reporting this. Here is the right issue. I renamed the issue to reflect that it collects multiple monitors/TVs. I had a look at the user's manual: Your UE32J5150 is supposed to support 720p @ 50 Hz, so the problem is in the core.

Just for clarity: Did you already try these steps? And if not, can you try them?
https://github.com/MJoergen/C64MEGA65#hdmi-compatibility

ChristianWeyer, 2022-07-02:
1. Yes, I am using version 3 of the core.
2. Yes, I tried all these steps - to no avail.

Thanks @sy2002 ! :-)

iFlash, 2022-07-02:
No, it’s connected with HDMI.

PS: Sent from mobile device - please excuse typos

sy2002, 2022-07-03:
@ChristianWeyer Can you check if this core (ZIP attached) is showing any picture on your screen?
[coretest.zip](https://github.com/MJoergen/C64MEGA65/files/9034720/coretest.zip)

ChristianWeyer, 2022-07-03:
Hin @sy2002 - nope, no display/picture with this core.

sy2002, 2022-07-03:
@ChristianWeyer Thank you for trying.

sy2002, 2022-07-18:
Thank you @jeffreyvanderschilden . This whole thing is a really interesting puzzle for us. The difference between the MEGA65's HDMI output and the C64 core's (and all MiSTer2MEGA65 enabled cores) is:

MEGA65 outputs PAL and NTSC resolutions via HDMI, i.e. 720 x 576 pixels for PAL at 50 Hz and 720 x 480 for NTSC at 60 Hz, 4:3, while the C64 core (and MiSTer2MEGA65) outputs 720p, i.e. 1280x720 pixels at 50 Hz (or 60 Hz), 16:9 but for the C64 with a black border left/right so that the C64 is still pixel perfect 4:3.

The puzzling thing is: We use the same HDMI logic as the MEGA65, but still something in our timing and/or HDMI data packages seems to be off, so that some Monitors/Devices work and some don't.

I guess to solve that, we need access to a HDMI signal analyzer.

0ddjob, 2022-10-18:
If it helps ... I'm one of those who couldn't use the new core as nothing appeared on several different HDMI TVs ... ones at home and in several hotels (I take my MEGA with me on work road trips).

But ... I've finally found a HDMI capture device that works with the MEGA65 ... one of those cheapo HDMI-to-USB units.  

And ... surprise!  The C64 RC3 core is finally working for me!  My MEGA is connected to my Mac Mini via the HDMI converter dongle and I'm using Quicktime Player to "record new movie" ... and I can now see video output.  

Brett.

0ddjob, 2022-10-18:
If it helps ... I'm one of those who couldn't use the new core as nothing appeared on several different HDMI TVs ... ones at home and in several hotels (I take my MEGA with me on work road trips).

But ... I've finally found a HDMI capture device that works with the MEGA65 ... one of those cheapo HDMI-to-USB units. I'd been trying on both Mac and Win10 with an Elgato HD60S+ with little success, and none at all with a Razer Ripsaw (unsupported device) ... I'm thinking these 4K 60fps devices weren't designed with 720x576 in mind.

And ... surprise! The C64 RC3 core is finally working for me! My MEGA (PAL 50Hz) is connected to my Mac mini via the HDMI converter dongle and I'm using Quicktime Player to "record new movie" ... and I can now see video output.

I tried directly to the hotel TV that the Mac mini is connected to (an LG model) ... no output (as usual for me).

Brett.

0ddjob, 2022-10-18:
Example ...
[image]

sy2002, 2022-10-18:
@0ddjob Thank you for sharing - this is very interesting. It triggers some thoughts. Let me soon share with you a test version that might (hopefully) work without the HDMI converter.

0ddjob, 2022-10-19:
> @0ddjob Thank you for sharing - this is very interesting. It triggers some thoughts. Let me soon share with you a test version that might (hopefully) work without the HDMI converter.

More than happy to give it a test when it's ready. 
/Brett

sy2002, 2022-11-06:
@jeffreyvanderschilden and @0ddjob : Can you please try if the new HDMI 4:3 mode that does 576p instead of 720p fixes the problems you described? Here is an unreleased Beta version of the core: 
[C64MEGA65-HDMI43.zip](https://github.com/MJoergen/C64MEGA65/files/9947093/C64MEGA65-HDMI43.zip)
For switching to the HDMI 4:3 mode, select the menu item "HDMI: 4:3 mode"
Make sure that you read [this](https://github.com/MJoergen/C64MEGA65/tree/develop#config-file) before trying the core.
Known bug: In 4:3 mode, the file-browser's first few columns are unreadable; we will fix that soon, this test is only about gauging if the 4:3 mode fixes the challenges you described.

sy2002, 2022-11-06:
@ChristianWeyer and @iFlash and @grim-fandango please see above message and experimental core.

jeffreyvanderschilden, 2022-11-07:
@sy2002 Thanks for the effort. Sadly black screen still.

Double checked settings by removing the capture device and hooking up to the monitor directly. Also tried different combinations with the DVI option on/off and force 60hz while in 4:3 mode. All combinations work fine directly on the monitor, while nothing works for the capture device.

Switching back to MEGA65 core all is fine again with the capture device.

sy2002, 2022-11-07:
@jeffreyvanderschilden Thank you for trying! This is really strange as the 4:3 mode is the same resolution as the MEGA65 outputs. Need to revisit. So 4:3 "pure" and 4:3 + DVI and Force 60 does not work. Can/did you also try DVI and Force 60 for the "non 4:3"?

jeffreyvanderschilden, 2022-11-08:
@sy2002 yes also tried that combination for sure.  It's funny that 576p is not officially supported on the capture device, but it takes the MEGA65 576p just fine. That said it definitely does support 720p officially, so the original non 4:3 mode of the C64 core should normally have been picked up I'd say. Probably a different issue unrelated to resolution I'd guess.

sy2002, 2023-01-16:
@grim-fandango and @ChristianWeyer and @iFlash and @0ddjob and @jeffreyvanderschilden : We found an incompatibility, where the core was not compliant to 4.2.7 of the HDMI specification 1.4b. We fixed it and now the HDMI output works with more monitors, frame grabbers, switches, etc. than ever before. Please test the attached core [C64M65-WIP-V5-A7.zip](https://github.com/MJoergen/C64MEGA65/files/10425145/C64M65-WIP-V5-A7.zip) and report back here on in Discord in the #other-cores channel:
 
https://discord.com/channels/719326990221574164/794775503818588200/1064498334515068958

The attached experimental core for the upcoming version 5 of the C64 core fixes this very issue. Would be cool, if all of you who experienced this problem could test the attached core and share feedback. Disclaimer: This is an alpha version and therefore not meant to be your "production C64 core". Having said that: It is pretty stable and runs more demos than the official release 4, for example "All Hallow's Eve".

tmilker, 2023-01-21:
@sy2002 The WIP-V5-A7 core worked to fix the No Signal on my Elgato capture card.  Thanks!

sy2002, 2023-01-22:
Thank you for your feedback @tmilker 👍🏻

sy2002, 2023-01-25:
Additionally to the positive feedback from tmilker (https://github.com/MJoergen/C64MEGA65/issues/4#issuecomment-1399356955), we got feedback from the following Discord users that the [Alpha 7 for Version 5](https://github.com/MJoergen/C64MEGA65/files/10425145/C64M65-WIP-V5-A7.zip) core fixes the HDMI issue(s) they had: KiDra#7570, AmokPhaze101#8641 and Mike351#9399. And additionally to the Discord users, @MJoergen confirmed that with the fix his HDMI switch, that formerly did not work, now works.

**Bottomline: Five positive feedbacks. We consider this issue as fixed.**

The fix will be delivered in Version 5 (currently Version 4 is the newest version). So we leave the issue open with the tag "fixed-in-develop-branch" and we will close it as soon as Version 5 is released.

P.S. This issue is independent from this audio issue: https://github.com/MJoergen/C64MEGA65/issues/13

**IMPORTANT: DO NOT ADD MORE COMMENTS TO THIS ISSUE.** It is meant to be a "read-only" issue from now on.

If you experience HDMI problems, please [read the FAQ](https://github.com/MJoergen/C64MEGA65/blob/master/FAQ.md), try [Alpha 7 for Version 5](https://github.com/MJoergen/C64MEGA65/files/10425145/C64M65-WIP-V5-A7.zip) and if the issue persists please [OPEN A NEW ISSUE](https://github.com/MJoergen/C64MEGA65/issues/new/choose) here on GitHub.

sy2002, 2023-06-02:
Part of version 5

closed 2023-06-02 by sy2002