MJoergen/C64MEGA65 #5: Screen is shifted right, down since new MEGA65 core 180

Turboschlumpf, 2022-05-13:
Hi,

since I use the new MEGA64 core 180 the screen of the C64 is shifted to the right and down with the latest C64 core. I heard that also the Freezer menu is sifted since the new core so the new Freezer 1.5 must be used with this core. Does this need to be fixed or can this be configured anywhere?

Regards

sy2002, 2022-05-13:
Hi @Turboschlumpf . Great name :-)
Are you talking about VGA or HDMI?

Turboschlumpf, 2022-05-13:
Hi sy2002,

ah, yes, I missed to say that. I talk about VGA. I can take a photo, when I'm back home.
Thanks for the quick answer.

Regards

Turboschlumpf, 2022-05-13:
Hi sy2002,
the Monitor I'm using is a LG TV. It is not 15KHz capable. The MEGA65 cor 180 is centered.
[image]
Attached a quick photo of the TV screen.

Regards

sy2002, 2022-05-15:
@Turboschlumpf   Thank you for the clarification. Try the auto-adjust of your Monitor, please. Each Monitor capable of VGA should have an auto-adjust. The C64 core outputs a slightly different signal than the MEGA65 core: The MEGA65 core outputs a "clean" PAL or NTSC signal. The Commodore 64 though, was not as "clean" back in the days and this is why also our implementation (which is a MiSTer port) tries to be as close at the original C64 as possible: The C64 and the VICII worked with very odd frequencies, for example see here: https://codebase64.org/doku.php?id=base:pixel_aspect_ratio

Long story short: The Auto-Adjust of your monitor (a menu item which normally is hidden in monitor's menus and only becomes visible when you are actually in a VGA mode) should solve it. Please confirm if this is indeed so.

(And you need to re-auto adjust each time you switch between C64 and MEGA65).

Turboschlumpf, 2022-05-16:
Hi sy2002, thanks for making it clear. Autoadjust is not easy to reach, as the buttons are on the rear right on this TV. That's not so nice. I probably need to use another monitor then.

Regards

sy2002, 2022-05-17:
Thank you @Turboschlumpf for your feedback. I will close this issue now, because I assume that Autoadjust is the solution. Please re-open, if not.