MJoergen/C64MEGA65 #6: No bootup after flashing on the MEGA65

oe3gwu, 2022-06-02:
I flased the ROM, and I was able to start the Core once. Then I turned the MEGA65 off and on again, and since then the 2 LED's blink blue in sequence. Also a Core reflash doesnt work. Also no display output.

oe3gwu, 2022-06-02:
I found out, if I remove the SD Card when booting the core, the system comes up. When I insert it, and boot up, the LEDs flash in sequence. Why?

grim-fandango, 2022-06-02:
Did you try the reset fix core here: https://github.com/MJoergen/C64MEGA65/blob/dev-dvi/bin/Version%201/R3/C64MEGA65-1-R3-resetfix-100MHz.cor ? This fixes the 'ambulance lights' issue

oe3gwu, 2022-06-02:
I downloaded the core here:

https://sy2002.github.io/m65cores/

It refers to: https://files.mega65.org/?id=896a012f-59e4-456c-b91f-7e989b958241

I have no idea.

I will try this core tomorrow.

oe3gwu, 2022-06-03:
Thanks this solved the issue. Works perfect now!

closed 2022-06-03 by oe3gwu