MJoergen/C64MEGA65 #7: D64 variations not supported

glbyers, 2022-07-09:
The D64 disk format has several variations, including 35 track with an error map, 40 track & 40 track with error map (see https://wiki.retrograde.dk/doc:cbm:disk:image:d64#variations_on_the_d64_layout). This limits the ability to support non-cracked originals. Note that this will likely not be a problem once G64 is supported, but it would be nice to support the extended D64 formats for completeness of compatibility.

sy2002, 2022-08-17:
@glbyers Thank you for your input (and sorry for the delayed feedback - I missed this issue):

**Yes you are right - good point. I added it to our roadmap, which [you can find here (Link)](https://github.com/MJoergen/C64MEGA65/blob/develop/ROADMAP.md) and G64 support is also on the roadmap.**

Can you send me some example D64 files in these formats to my private email `code at sy2002.de` and/or approach me via a PM in the MEGA65 discord? I cannot commit to any delivery date at this moment but I fully endorse the idea of supporting all D64 variations so I would like to put some examples on my "stack" so that when I find time to address this issue I have everything I need to proceed (without lengthy internet research).

sy2002, 2022-10-29:
Version 4 which is scheduled "before Christmas" this year will support 40 track D64 files (standard format, without error bytes), so that you can for example run Censor design's demos "We come in peace" and "Star Wars". I tested both and they work like a charme.

@glbyers in case you want to beta-test, attached you find a ZIP with the core which supports 35 and 40 tracks: [C64MEGA65-40-Tracks.zip](https://github.com/MJoergen/C64MEGA65/files/9893788/C64MEGA65-40-Tracks.zip)

sy2002, 2022-12-20:
Progress Update:

* 35 and 40 track D64 files (without error bytes) are supported with [Version 4](https://github.com/MJoergen/C64MEGA65/releases/tag/V4).

* Disk images with an error map are not supported, yet

This is why we keep this issue open.

sy2002, 2024-10-30:
@paich64 and @Kugelblitz360 do you gentlemen have examples of D64 disk images that are exactly 175,531 and 197,376 bytes in size which are the "with error maps" D64 file formats? This here would be a rather easy thing to support, if I were able to test it (and I I knew, if this feature is needed at all).

Currently (see above in this issue), since Version 4 the core supports 35 and 40 track D64 files.

Kugelblitz360, 2024-10-30:
http://unusedino.de/ec64/technical/formats/d64.html
has a full description.
https://style64.org/dirmaster
should support those files, but I can't find anything in the docs and won't be able to try before Sunday night.

sy2002, 2024-10-30:
Thanks @Kugelblitz360 . The question circles a bit around: Is this needed at all. I mean D64 with error maps (which is completely independent from G64). Is there any relevant use "in the market out there" of this stuff. Otherwise we could also say: We have enough other good ideas so let's scratch this idea/issue. Honestly, I just picked it for 5.3 due to being relatively easy to do and not changing the config file (see Skype).

sy2002, 2024-10-30:
Hmm... Are you sure that if you transfer personal disks to your PC, then then people are not using G64 instead of D64?

Kugelblitz360, 2024-10-30:
I think we can summarize to:
* I see no archive of d64 with error maps games that is actually in use. Stuff floating on the net does not use this format
* None of the edit tools for d64 floating around actually modifies the error map and many don't even show it
* VICE does support these in TDE for d64 and there are discussions on Lemon that contain files for games from Windham Classics that apparently will work as error mapped D64s. https://www.lemon64.com/forum/viewtopic.php?p=936275
* The only other use case I have seen are people reading their disks into error mapped .d64 to see whether the data is "good" and there are tools to repair "weak" disks by taking the input of several .d64s and comparing the error maps.
In the end these are all fringe cases. Vice needs to jump through hoops to simulate the effects of some of the errors and for every one of these cases, using a .g64 would be more straightforward.

I think we can live without error mapped d64s pretty well. Getting NTSC up and running would be 100x more useful, for example.

Kugelblitz360, 2024-10-30:
Ah, my VPN works (slowly), can go through my collection now. There is an Action Replay Tool disk that needs the error map (attached).
Gamebase V19 has quite a lot of d64 files with errormaps. Also in the Tip are examples, the full list would have to wait until Sunday night.
[d64errors.zip](https://github.com/user-attachments/files/17579327/d64errors.zip)

sy2002, 2024-11-03:
Thank you @Kugelblitz360 ! This is super valuable input as it shows that the feature is worth the effort as there are D64s "in the wild" that can make use of the feature. When you talk about Gamebase V19 (which I do not know), do you mean this here:

https://archive.org/details/gb64v19

Kugelblitz360, 2024-11-03:
a) Yes, that is the one.
b) DirMaster/Style does show the error map when you show the BAM. You can edit errors by right clicking a sector,
[image]
![image](https://github.com/user-attachments/assets/f6072ebe-11f9-4c9f-bb70-9ab90257a72
[c64errormaps.zip](https://github.com/user-attachments/files/17612358/c64errormaps.zip)
c) Included some more files in ZIP, including 2 40 Track images with Error Map. "Orange_Line_2*" will crash VICE when not loaded with True Drive Emulation...