MJoergen/C64MEGA65 #13: HDMI: Random micro-cuts of sound

sy2002, 2023-01-22:
User Tonton#0297 reported on Discord: "Random micro-cuts of the sound through the hdmi port. No issue with the audio jack of the Mega65" [Discord Link here](https://discord.com/channels/719326990221574164/794775503818588200/1066405370555289660).

@ambl5 and @MJoergen

amb5l, 2023-01-22:
This has to be a small but cumulative delta between perfect and actual frequencies of the pixel and PCM clocks. Would one of you be able to summarise the MMCM recipe (ref clock, multiplier,  divider, output divider) for each?

Is the PCM clocking done in the "clock enable" way? If so there may be a tiny but cumulative error in the counting logic. It should be quite easy to create a small/fast simulation focused on looking for this.

kaloun34, 2023-01-22:
hello
This issue is random but can happen quickly 

https://user-images.githubusercontent.com/54940754/213914777-250c543d-5b88-46bf-91c4-8862b5332486.mp4

sy2002, 2023-01-23:
The video link does not seem to work on GitHub, here is a link to Discord that does not seem to need any login to watch. The glitch seems to appear after 7 seconds where the audio goes away and then comes back:
https://discord.com/channels/719326990221574164/794775503818588200/1066690068531658813

sy2002, 2023-01-23:
Tonton confirmed that the problem occurs in our old (pre-audio refactoring) version of the core V4 as well as in the new alpha version A7 for version 5 after @MJoergen refactored some stuff - so it looks like it was not the audio refactoring and we can use the refactored version in the `develop` branch to investigate further.

@amb5l about your questions: 

Here is the clocking for the Commodore 64:
https://github.com/MJoergen/C64MEGA65/blob/develop/CORE/vhdl/clk.vhd

As you can see there, we are - when the mode "HDMI: Flicker-fix" is ON slowing down the whole C64 system to make sure that the framerate is much closer to 50 Hz(*) and therefore we don't have tearing/flickering effects while scrolling:
https://github.com/MJoergen/C64MEGA65/blob/develop/CORE/vhdl/clk.vhd

(*) re 50 Hz: The core is a PAL-only core at this moment, therefore 50 Hz.

The MiSTer2MEGA65 framework is then translating the C64 output to HDMI using this clocking:
https://github.com/MJoergen/C64MEGA65/blob/develop/M2M/vhdl/clk_m2m.vhd

As you can see there, everything is pretty plain vanilla the frequency multipliers and dividers for example for 74.25 MHz for 720p (50 Hz or 60 Hz) and 5x74.25 MHz = 371.25 MHz for TMDS result in "nice" values. For example:
```
f_VCO = (100 MHz / 5) x 37.125 = 742.5 MHz
742.5 / 2 = 371.25 MHz
742.5 / 10 = 74.25
```

About your question "s the PCM clocking done in the "clock enable" way?":

Yes, it is generated from the HDMI clock via a clock enable:

```vhdl
   -- ACR packet rate should be 128fs/N = 1kHz
   p_pcm_acr : process (hdmi_clk_i)
   begin
      if rising_edge(hdmi_clk_i) then
         if pcm_clken = '1' then
            -- Generate 1KHz ACR pulse train from 48 kHz
         if pcm_acr_counter /= (pcm_acr_cnt_range - 1) then
            pcm_acr_counter <= pcm_acr_counter + 1;
            pcm_acr <= '0';
         else
            pcm_acr <= '1';
            pcm_acr_counter <= 0;
         end if;
      end if;
      end if;
   end process p_pcm_acr;
```

The clock enable `pcm_clken` is generated like this:

```vhdl
   -- Generate PCM sample rate (48 kHz)
   i_clk_synthetic_enable : entity work.clk_synthetic_enable
      port map (
         clk_i       => hdmi_clk_i,
         src_speed_i => hdmi_video_mode.CLK_KHZ * 1000,
         dst_speed_i => HDMI_PCM_SAMPLING,
         enable_o    => pcm_clken
      ); -- i_clk_synthetic_enable
 ```
 
 And the `clk_synthetic_enable` component works like this:
 https://github.com/MJoergen/C64MEGA65/blob/develop/M2M/vhdl/av_pipeline/clk_synthetic_enable.vhd

sy2002, 2023-01-23:
@kaloun34 

Which HDMI mode are you using? (Which resolution, Hz) AND: Do you have HDMI: Flicker-free on or off when this happens?

kaloun34, 2023-01-23:
[image]

sy2002, 2023-02-12:
@kaloun34 Can you please test this core? [hdmi_tpg_mega65r3.zip](https://github.com/MJoergen/C64MEGA65/files/10717474/hdmi_tpg_mega65r3.zip)

It is an HDMI test core. @amb5l built it and it uses an advanced version of the HDMI engine that the C64 core is using. The test core shows a test image and plays a test sound.

**Are the random micro-cuts still audible** on your monitor with this test core?

When the core starts, please press MEGA65's <kbd>RETURN</kbd> key 8 times so that you are in 576p @ 50 Hz. When finished testing this mode, press the <kbd>RETURN</kbd> key two more times (10 times overall) to enter the mode 720p @ 50 Hz. These two modes are used by the C64 core.

sy2002, 2023-02-12:
Just in case the `.bit` above is not sufficient, here is a `.cor`: [hdmi_tpg_mega65r3.zip](https://github.com/MJoergen/C64MEGA65/files/10717511/hdmi_tpg_mega65r3.zip)

kaloun34, 2023-02-13:
Hello
Random micro-cuts are not audible with this Core test, but I would need continuous audio to be sure.

amb5l, 2023-02-17:
@kaloun34 here is an updated test pattern generator with a continuous audio output feature (a ZIP of a bit file). To make the audio tone continuous, hold down the **INST/DEL** key on the MEGA65 keyboard.

[hdmi_tpg_mega65r3.zip](https://github.com/MJoergen/C64MEGA65/files/10767471/hdmi_tpg_mega65r3.zip)

kaloun34, 2023-02-18:
Thank you, I will test
I have to remember how to flash with a bitstream file first

sy2002, 2023-02-18:
@kaloun34 Here is a `.cor` version of the bitstream: [hdmi_tpg_mega65r3-cor.zip](https://github.com/MJoergen/C64MEGA65/files/10773704/hdmi_tpg_mega65r3-cor.zip)

(You can only use `.bit` of you own a JTAG cable. `.bin` is the native format that tools like Vivado or also MEGA65's own `m65` tool can work with and `.cor` is the format that the built-in MEGA65 core-flasher can work with. If you want to learn more: https://github.com/MEGA65/mega65-tools - but for now the above-mentioned `.cor` should do the job.)

kaloun34, 2023-02-18:
> @kaloun34 here is an updated test pattern generator with a continuous audio output feature (a ZIP of a bit file). To make the audio tone continuous, hold down the **INST/DEL** key on the MEGA65 [...]

I tested it and it seems to be good.
That sound for 30 minutes is hellish  :)

sy2002, 2023-02-18:
@kaloun34 Sounds great!

P.S. Is this a joke or did you really test for 30mins?

sy2002, 2023-02-18:
In Discord you wrote that you tested 576p @ 50 Hz - just for being on the safe side and just because the screenshot that you posted above (scroll up) shows 720p @ 50 Hz: Did you test both resolutions as described [here](https://github.com/MJoergen/C64MEGA65/issues/13#issuecomment-1427150669)?

kaloun34, 2023-02-19:
> @kaloun34 Sounds great!
> 
> P.S. Is this a joke or did you really test for 30mins?

Yes, I tested 15 minutes in each of the two resolutions  576p and 720p

sy2002, 2023-02-19:
@kaloun34 Excellent work! Thank you. Then we know what next steps we need to perform to make the C64 Core (and the MiSTer3MEGA65 Framework) HDMI sound output better and more stable.

Due to personal time constraints: This might take a few months - but it will eventually happen. Please bear with us :-)

sy2002, 2023-05-10:
@kaloun34 Unfortunately this bugfix will not make it to Version 5 (ETA "before summer"/end of June).

MJoergen, 2023-11-14:
@kaloun34 Could you please try the attached bit file? With any luck, this will have fixed the audio micro-cuts.
[C64MEGA65_V51_HDMI_FIX.zip](https://github.com/MJoergen/C64MEGA65/files/13357579/C64MEGA65_V51_HDMI_FIX.zip)

sy2002, 2023-11-15:
@kaloun34 Just in case you don't have a JTAG adaptor and need a `.cor` instead of a `.bit` file to test, here you hare: [COR-C64MEGA65_V51_HDMI_FIX.zip](https://github.com/MJoergen/C64MEGA65/files/13361794/COR-C64MEGA65_V51_HDMI_FIX.zip)

kaloun34, 2023-11-15:
Hello,

 

The problem seems to have been solved.

But I'll have to test it longer

 

Thank you so much

sy2002, 2023-11-15:
@kaloun34 This is amazing feedback! Awesome! Thank you :-)

sy2002, 2024-01-13:
[This comment](https://github.com/MJoergen/C64MEGA65/issues/116#issuecomment-1880157533) confirms that the V5.1 `develop` branch of V5.1 using Tyto 2 is fixing this. In case we need to roll-back to Tyto 1 due to https://github.com/MJoergen/C64MEGA65/issues/116 then this issue here would slip to V6.

sy2002, 2024-01-14:
@paich64 and @kaloun34 : Sorry for asking again to test, but as it looks like this time the last time: Is "Random micro-cuts-of-sounds" still fixed with Alpha 9: [C64MEGA65-V5.1A9.zip](https://github.com/MJoergen/C64MEGA65/files/13931765/C64MEGA65-V5.1A9.zip) ?

If so, we could then close this issue.

paich64, 2024-01-14:
I asked directly @kaloun34 to perform the test. we'll wait for hist feedback.

paich64, 2024-01-16:
@sy2002 I have just talked to @tonton and he just confirmed to me that after having tested V5A9, the microcut issue is still fixed with this last core version. I guess you can close this issue :)

sy2002, 2024-01-16:
Thank you @kaloun34 and @paich64 these are great news!

commit 3030cb7d (2023-11-16): Fix HDMI audio microcuts

This should hopefully fix #13
closed 2024-01-16 by sy2002