MJoergen/C64MEGA65 #15: SID player

zildac, 2023-02-12:
Would you consider adding a SID player to the HELP menu to allow playing of SID files from the SDCARD? This would save having to load a native 64 SID player and then mount an image full of SIDs.

sy2002, 2023-02-12:
@zildac This is indeed a nice idea but unfortunately far beyond the scope of our project. At least for the next 1-2 years or so. Here is an idea for a workaround that might give you still what you are looking for:

1. Download a program like this:
https://csdb.dk/release/?id=161475

2. Convert it to a cartridge (.crt) by googling `convert c64 prg into crt` and then experimenting a bit

3. Wait until Version 5 of the core is released - hopefully before or in summer 2023: Version 5 of the core will feature cartridges: Physical cartridges that you stick into your MEGA65's expansion port - but also software cartridges (`.crt`). If you now mount such a cartridge, in the Help menu, then you have nearly exactly what you wanted. Another way would be the EasyFlash hardware cartridge and then flash the SidPlayer there and then stick it into the MEGA65. But as said: This will need Version 5 which is currently in development.

I am grateful for your idea. But I will close it for the above-mentioned reason and hopefully the workaround will be feasible in future.

closed 2023-02-12 by sy2002

zildac, 2023-02-13:
Understood, and no problem. Thanks for taking a look 👍