MJoergen/C64MEGA65 #20: SIMCRT: Modern Ocean cartridges do not work

sy2002, 2023-04-25:
Ocean is an important format. Back in the retro days it was used very often.

Skype Message from 4/24/23 from sy2002 to MJoergen reads: "Attached you find a ZIP. It is unique in the sense that it contains the same game released for 4 different cartridge types: EasyFlash, GMOD2, gs54 and Ocean.

The latter cartrige type - Ocean - crashes after the "In loving memory" screen. The other cartridge types work."

You find the ZIP attached to this very Skype message

MJoergen, 2023-04-26:
1. I've reproduced the crash on my machine.
2. I've verified that the cartridge works in VICE.
3. Just before the time of crash, the debug output from VICE is:
```
.837E 265 050   40056017  84 C6       STY $C6    051000f9 
.8380 265 053   40056020  20 3C AF    JSR $AF3C  051000f9 
.AF3C 265 059   40056026  A4 21       LDY $21    051000f7 
.AF3E 266 012   40056042  A6 20       LDX $20    051001f7  
```
4. On the MEGA65, the contents at $AF3C is $00, and so the CPU executes a BRK instruction, instead of the LDY $21.
5. At the same time, GAME=0 and EXROM=0, and the CPU I/O register contains $36.
6. At the same time, ROMH is asserted, and BANK_HI = 0 (as well as BANK_LO = 0).
7. Now, output from LO_RAM is $3114, while HI_RAM output is $0000.

I would expect the HI_RAM output to be identical to LO_RAM output, since BANK_HI = BANK_LO. Whether this is the root cause of the problem is so far unknown.

MJoergen, 2023-04-26:
Examining the CRT file, it is of type 5 (Ocean Type 1).

The CRT file contains a total of 64 "chips", all of them of length 8kB, and starting at address $8000.

MJoergen, 2023-04-27:
Extra debug output from VICE shows the following:
At startup, the CPU I/O register contains 7 (reset value).
At time of executing instruction at $AF3C, the CPU I/O register contains 6, i.e. LORAM is cleared. In this mode address $AF3C is taken from ROM_HI.

A comment in MiSTer's cartridge.v suggests that (for Ocean Type 1) ROM_HI is a mirror image of ROM_LO.

MJoergen, 2023-04-27:
Searching the CRT file for the byte sequence $A4, $21, $A6, $20, which is the sequence appearing at $AF3C, this particular byte sequence appears twice, at address $76BED and $76BFF. Neither of these two addresses are 4-byte aligned, which I would expect, given that the address $AF3C is 4-byte aligned.

MJoergen, 2023-04-27:
The following shows the sequence of changes to the CPU I/O register.

- The second-to-last value is contains the CPU I/O register in bits 2-0, i.e. bit 2 is CHAREN, bit 1 is HIRAM, and bit 0 is LORAM.
- The last column is the ROM bank number (same for both ROM_HI and ROM_LO).
- Column 4 is the clock cycle counter, at roughly 1 MHz. So approximately equal to the time in micro-seconds.

```
.FCE2 000 006          6  A2 FF       LDX #$FF    00000af3 0f 00
.68AF 297 021      18732  E8          INX         0d3668fd 0f 36
.68AF 222 021     131943  E8          INX         a03700fd 0f 37
.0DB6 002 030     727428  8C 0E DC    STY $DC0E   350000fb 0d 37
.110E 007 056   34064345  EA          NOP         3700fffd 0f 37
.6870 000 014   34103174  AE 02 9F    LDX $9F02   00ffffff 0f 00
.68AF 000 038   34103198  E8          INX         7f38fffd 0f 38
.68AF 237 038   34216409  E8          INX         a03900fd 0f 39
.6886 162 034   34329616  AE 01 9F    LDX $9F01   003a00ff 0f 00
.68AF 162 058   34329640  E8          INX         043a00fd 0f 3a
.68AF 087 058   34442851  E8          INX         a03b00fd 0f 3b
.68AF 012 058   34556062  E8          INX         a03c00fd 0f 3c
.68A9 250 004   34669282  4C 00 04    JMP $0400   363d00ff 0e 3c
.3118 311 029   34673150  A9 00       LDA #$00    375005fd 0f 3c
.311D 311 035   34673156  BC 00 9F    LDY $9F00,X 005005fd 0f 00
.3123 311 043   34673164  60          RTS         00502cfd 0f 2c
.3103 142 008   34682138  BD 00 9D    LDA $9D00,X 0009fff9 0f 00
.3113 142 030   34682160  60          RTS         870904f9 0f 04
.30F9 306 007   34751437  60          RTS         360900fb 0e 04
.3118 233 025   34825480  A9 00       LDA #$00    374f00fd 0f 04
.311D 233 031   34825486  BC 00 9F    LDY $9F00,X 004f00fd 0f 00
.3123 233 039   34825494  60          RTS         004f2dfd 0f 2d
.045E 233 050   34825505  8C F2 05    STY $05F2   004f2dff 0e 2d
.30FE 097 006   34856205  A9 00       LDA #$00    376b12fb 0f 2d
.3103 097 012   34856211  BD 00 9D    LDA $9D00,X 006b12fb 0f 00
.3113 097 034   34856233  60          RTS         9e6b32fb 0f 32
.22BA 251 014   34865915  AD 10 D0    LDA $D010   36a605f2 0e 32
.22C3 252 044   34866008  4C D5 22    JMP $22D5   370006f3 0f 32
.22BA 255 012   34885821  AD 10 D0    LDA $D010   36a605f2 0e 32
.22C3 256 042   34885914  4C D5 22    JMP $22D5   370006f3 0f 32
.22BA 255 014   34905479  AD 10 D0    LDA $D010   36a605f2 0e 32
.22C3 256 044   34905572  4C D5 22    JMP $22D5   370006f3 0f 32
.22BA 255 017   34925138  AD 10 D0    LDA $D010   36a605f2 0e 32
.22C3 256 047   34925231  4C D5 22    JMP $22D5   370006f3 0f 32
.22BA 255 017   34944794  AD 10 D0    LDA $D010   36a605f2 0e 32
.22C3 256 047   34944887  4C D5 22    JMP $22D5   370006f3 0f 32
.30F9 043 051   34951128  60          RTS         360000fd 0e 32
.30FE 087 008   34953857  A9 00       LDA #$00    3703fffb 0f 32
.3103 087 014   34953863  BD 00 9D    LDA $9D00,X 0003fffb 0f 00
.3113 087 036   34953885  60          RTS         9e032cfb 0f 2c
.30F9 219 035   34962200  60          RTS         360300fd 0e 2c
.30FE 261 030   39701937  A9 00       LDA #$00    376a1afb 0f 2c
.3103 261 036   39701943  BD 00 9D    LDA $9D00,X 006a1afb 0f 00
.3113 261 058   39701965  60          RTS         8e6a3bfb 0f 3b
.22BA 251 012   39720945  AD 10 D0    LDA $D010   36a605f2 0e 3b
.22C3 252 042   39721038  4C D5 22    JMP $22D5   370006f3 0f 3b
.22BA 255 015   39740856  AD 10 D0    LDA $D010   36a605f2 0e 3b
.22C3 256 045   39740949  4C D5 22    JMP $22D5   370006f3 0f 3b
.22BA 255 014   39760511  AD 10 D0    LDA $D010   36a605f2 0e 3b
.22C3 256 044   39760604  4C D5 22    JMP $22D5   370006f3 0f 3b
.22BA 255 012   39780165  AD 10 D0    LDA $D010   36a605f4 0e 3b
.22C3 256 042   39780258  4C D5 22    JMP $22D5   370006f5 0f 3b
.22BA 255 017   39799826  AD 10 D0    LDA $D010   36a605f2 0e 3b
.22C3 256 047   39799919  4C D5 22    JMP $22D5   370006f3 0f 3b
.22BA 255 014   39819479  AD 10 D0    LDA $D010   36a605f4 0e 3b
.22C3 256 044   39819572  4C D5 22    JMP $22D5   370006f5 0f 3b
.22BA 255 013   39839134  AD 10 D0    LDA $D010   36a605f2 0e 3b
.22C3 256 043   39839227  4C D5 22    JMP $22D5   370006f3 0f 3b
.22BA 255 016   39858793  AD 10 D0    LDA $D010   36a605f4 0e 3b
.22C3 256 046   39858886  4C D5 22    JMP $22D5   370006f5 0f 3b
.22BA 255 013   39878446  AD 10 D0    LDA $D010   36a605f2 0e 3b
.22C3 256 043   39878539  4C D5 22    JMP $22D5   370006f3 0f 3b
.30F9 139 051   39890832  60          RTS         360000fd 0e 3b
.30FE 180 060   39893424  A9 00       LDA #$00    3706fff9 0f 3b
.3103 181 003   39893430  BD 00 9D    LDA $9D00,X 0006fff9 0f 00
.22BA 255 017   39898106  AD 10 D0    LDA $D010   36a605f4 0e 00
.22C3 256 047   39898199  4C D5 22    JMP $22D5   370006f5 0f 00
.22BA 255 015   39917760  AD 10 D0    LDA $D010   36a605f4 0e 00
.22C3 256 045   39917853  4C D5 22    JMP $22D5   370006f5 0f 00
.22BA 255 012   39937413  AD 10 D0    LDA $D010   36a605f4 0e 00
.22C3 256 042   39937506  4C D5 22    JMP $22D5   370006f5 0f 00
.22BA 255 013   39957070  AD 10 D0    LDA $D010   36a605f4 0e 00
.22C3 256 043   39957163  4C D5 22    JMP $22D5   370006f5 0f 00
.22BA 255 017   39976730  AD 10 D0    LDA $D010   36a605f4 0e 00
.22C3 256 047   39976823  4C D5 22    JMP $22D5   370006f5 0f 00
.30F9 233 040   39995023  60          RTS         360600fb 0e 00
.3003 235 036   39995145  60          RTS         300c01fb 08 00
.3008 115 034   40007239  58          CLI         360e01fb 0e 00
```

MJoergen, 2023-04-27:
I'm using this page [https://www.c64-wiki.com/wiki/Bank_Switching](https://www.c64-wiki.com/wiki/Bank_Switching) as reference for the ROM switching (PLA functionality).

MJoergen, 2023-04-27:
I believe the problem is that MiSTer sets GAME=0 and EXROM=0 for all Ocean Type 1 cartridges.
However, VICE is setting GAME=1 and EXROM=0.

In the Bank_Switching link above, with CPU I/O = 6, we compare lines 6 and 14, and see that the difference is precisely the bank $A000, which in MiSTer is mapped to ROMH, but in VICE is mapped to RAM.

MJoergen, 2023-04-27:
In VICE there is the following comment regarding Ocean cartridges:

```
      "Type A" cartridges work like this:

      - the hardware uses 16k game config, the same 8k ROM bank is mapped
        to both ROML and ROMH.
      - the software accesses banks 0-15 at ROML ($8000-$9FFF) and banks 16-31 at
        ROMH ($A000-$BFFF) (only 256k type)

      "Type B" cartridges work like this:

      - the hardware uses 8k game config, 8k ROM bank is mapped to ROML.
      - the software accesses banks 0-63 at ROML ($8000-$9FFF).
```

Here "16k game config" means EXROM=0 and GAME=0, while "8k game config" means GAME=1 and EXROM=0.

The game in question appears to be a "Type B" cartridge, but MiSTer only supports "Type A".

The way VICE distinguishes between these two types is purely based on the size of the cartridge:
- Type A : Up to 256k
- Type B : 512k

MJoergen, 2023-04-27:
Fixed in commit #b28b77c

closed 2023-04-27 by MJoergen

sy2002, 2023-04-27:
Just tested it. Confirmed.