MJoergen/C64MEGA65 #22: SIMCRT: Final Cartridge 3 does not work

sy2002, 2023-04-25:
If Final Cartridge 3 worked, we might not need to implement Jiffy DOS, because FC3 has a great fastloader. It also sports a debugger, i.e. we could in future more easily debug stuff.

I guess the not-working has to do with some lines such as NMI, IRQ, reset, etc. not being connected correctly. As this is a freezer module, it needs these lines.

Also have a look at cartridge.v as there is some "special treatmens" there:
https://github.com/MJoergen/C64_MiSTerMEGA65/blob/d7db318c97af237b3503fbac6858c05dce410bda/rtl/cartridge.v#L222

[Final_Cartridge_3_1988.zip](https://github.com/MJoergen/C64MEGA65/files/11320347/Final_Cartridge_3_1988.zip)

MJoergen, 2023-05-23:
Fixed in commit de4d570

sy2002, 2023-05-23:
I will close this as a Duplicate of https://github.com/MJoergen/C64MEGA65/issues/71 even though it technically isn't.

But right now, I would classify the Final Cartridge III as "not working": The Monitor is not working and I did not even do more intensive tests, if "the basics" such as the Monitor are not working. So I just assume that fixing above mentioned issue 71 will also fix more stuff.

closed 2023-06-02 by sy2002