MJoergen/C64MEGA65 #24: SIMCRT: Saving to cartridges

sy2002, 2023-04-25:
This would be a collaboration project because it needs The Shell to be able to save data per cartridge plus it would need some additional work in the CRT simulation. At this moment (4/25/23) I would classify this as an enhancement for a later Version 6, not for the currently WIP Version 5 - but let's see how it goes:

These games are known for not being able to save to simulated cartridges:

```
Sam's Journey cannot save to cartridge
Eye of the Beholder
Briley Witch Chronicles (v1.0).crt                                 => Trying to save to cart will fail (handled by the game)
Briley Witch Chronicles (v1.0.6).crt                               => Trying to save to cart will fail (handled by the game)
```

paich64, 2023-04-25:
I will provide all the official cartridge files soon.

MJoergen, 2023-05-09:
@sy2002 I agree that we should postpone this feature for a future release 6. The reason is that this will involve quite some development work; it's not a small change.

Now, the question remains how games handle this. If they give a sensible error message that is fine. But if a game just randomly crashes, then we should consider implementing some quick-fix or work-around, so games at least don't crash.