MJoergen/C64MEGA65 #25: SIMCRT: PacMan 25th Anniversary (EasyFlash) does not work

sy2002, 2023-04-25:
The game crashes / does not start after the intro.
You find the game in Skype in my message from 4/25/23.

MJoergen, 2023-04-27:
This cartridge works in VICE.

In VICE this cartridge starts in Ultimax mode (GAME=0, EXROM=1), but after some initialization, it switches over to 16k mode (GAME=0, EXROM=0). Later, it switches back and forth between basic mode (GAME=1, EXROM=1) and 16k mode.

Furthermore, it is executing code from $DFxx.

MJoergen, 2023-04-27:
Preliminary analysis shows that both in VICE and in hardware, at some point the code reaches $FFB1, which is the kernal LISTN function. In VICE the device number is 8.

In VICE this function returns via the RTS instruction at $EDAC. This instruction is never reached in hardware.

MJoergen, 2023-04-27:
Further analysis shows that both VICE and hardware reach the address $ED47. But VICE continues to the next instruction at $ED49, whereas in hardware we jump to $EDAD. The latter indicates "device not present".

MJoergen, 2023-05-04:
So, from address $2150 the program calls $FFB1 (kernal LISTN function) with A=0x08 at around 16.5 seconds after reset.
After that, at address $215B the program calls $FF93 (kernal SECND function) with A=0x6F.
Next, it repeatedly calls $FFA8 (kernal CIOUT function) with the values 0x4D, 0x2D, 0x57, 0x00, 0x05, 0x20. Then next with the values 0x0e, 0x43, 0x05, 0xAD, 0xA0, 0xFE, 0xA2, 0x03, 0xDD, etc.

Link: [https://www.commodore.ca/wp-content/uploads/2018/11/commodore_vic_1541_floppy_drive_users_manual.pdf](https://www.commodore.ca/wp-content/uploads/2018/11/commodore_vic_1541_floppy_drive_users_manual.pdf) (page 41)

This appears to be a "M-W" command sent to the 1541 drive, thus filling up from address 0x0500, i.e. "Buffer 2".

Then later the program sends "M-E" with the address 0x0500.

Later again, it sends a "M-R" requesting two bytes from address 0x0543.

The disassembly of the program uploaded to 0x0500 is:

```
0500   0E 43 05             ASL $0543
0503   AD A0 FE             LDA $FEA0
0506   A2 03                LDX #$03
0508   DD 36 05   L0508     CMP $0536,X
050B   F0 05                BEQ L0512
050D   CA                   DEX
050E   D0 F8                BNE L0508
0510   F0 19                BEQ L052B
0512   BD 39 05   L0512     LDA $0539,X
0515   8D 1F 05             STA $051F
0518   BD 3C 05             LDA $053C,X
051B   8D 20 05             STA $0520
051E   AD A4 FE             LDA $FEA4
0521   A2 03                LDX #$03
0523   DD 3F 05   L0523     CMP $053F,X
0526   F0 0B                BEQ L0533
0528   CA                   DEX
0529   D0 F8                BNE L0523
052B   AD C6 E5   L052B     LDA $E5C6
052E   C9 37                CMP #$37
0530   D0 01                BNE L0533
0532   CA                   DEX
0533   8E 44 05   L0533     STX $0544
0536   60                   RTS
0537   43 0D FF
053A   A4 C6 E9
053D   FE E5 A6
0540   38 46 48
0543   55
0544   00
0545   A2 00                LDX #$00
0547   AD 00 04             LDA $0400
054A   A8                   TAY
054B   29 0F                AND #$0F
054D   8E 00 18             STX $1800
0550   AA                   TAX
0551   BD 46 06             LDA $0646,X
0554   AE 00 18   L0554     LDX $1800
0557   F0 FB                BEQ L0554
0559   8D 00 18             STA $1800
055C   0A                   ASL A
055D   29 0F                AND #$0F
055F   8D 00 00             STA $0000
```

The bytes received are 0xAA and 0x00.

At address 0x2358 the first value is compared with 0xAA. At address 0x236D the second value is examined.

Address 0xFEA0 in the 1541 contains the low-byte of the address of "New" command.

Link: [https://g3sl.github.io/c1541rom.html](https://g3sl.github.io/c1541rom.html)

sy2002, 2023-05-04:
@MJoergen I remember Daniel Mantione telling us that there is a 1541 emulation built into the Easy Flash. My theory: So I assume that the game wants to "load" something using the "1541". What is supposed to happen (IMHO) is, that the cartridge should be in Ultimax mode during this moment so the vectors in the C64 Kernal which then call the "real 1541"'s routines are swapped out by vectors that point to the EasyFlash's emulated 1541 routines that load from the cartridge. So the problem might be as "easy" as making sure that we switch to Ultimax mode in the right moment?

I am adding AmokPhaze aka @paich64 to this discussion: @AmokPhaze can you please confirm if this `*.crt` cartridge runs on MiSTer?

I am also asking Daniel to have a look at this issue.

paich64, 2023-05-04:
I've just tested the EasyFlash cartridge on the last C64 mister core, and after the game loading phase, the screen remains black.

dmantione, 2023-05-04:
Hello,

I am tempted to say that what you are seeing can't be Disk2EasyFlash. Disk2EasyFlash implements the high-level KERNAL API by hooking the vectors at $0300. LISTN part of the low-level KERNAL API. The low-level API cannot be hooked. If LSTN is called from $2150, then this means the program is directly calling into the low-level API. This means it wants to talk to devices on the serial bus, not read the EasyFlash memory through the KERNAL API.

The game might be doing preparations for installing a fastloader into the 1541. The second half of the code shown is serial bus code.

Daniël

sy2002, 2023-05-04:
@dmantione Thank you for your feedback. This is interesting - a cartridge that installs a fastloarder? Maybe for saving highscores? Or maybe for using the 1541 als co-processor? It sounds strange borderline unrealistic, that a cartridge would only start, if the user has an 1541 attached AND switched on AND a disk inserted? :-)

@MJoergen Here is an experiment for you: Change this section in main.vhd:

```vhdl
   g_iec_drv_reset : for i in 0 to G_VDNUM - 1 generate
      iec_drives_reset(i) <= (not reset_core_n) or (not vdrives_mounted(i));
   end generate g_iec_drv_reset;
```

to

```vhdl
   g_iec_drv_reset : for i in 0 to G_VDNUM - 1 generate
      iec_drives_reset(i) <= (not vdrives_mounted(i));
   end generate g_iec_drv_reset;
```

[Mount this empty but correctly formatted D64](https://github.com/MJoergen/C64MEGA65/files/11401107/empty.zip) and then load the cartridge.

paich64, 2023-05-04:
I have just tested with the C64 mister core from the 24/02/2023, which do not include your commits, and the game fails exactly the same way.

paich64, 2023-05-04:
Just flashed my through hole EasyFlash cartridge and tested it on my Mega65 and my C64C 👍 

We exactly have the same behaviour on 
- Mister
- Mega C64 core with cartridge emulated
- Mega C64 core with through hole cartridge
- C64C with through hole cartridge.
More over there's a .d64 version of this same game.
It looks like to me this release is not clean.

sy2002, 2023-05-05:
@MJoergen Given that new information I would propose a CLOSE/WONTFIX if you're OK with that. And sorry for letting you hunt a ghost. I will ask @paich64 to confirm releases on MiSTer or on a real C64 before letting you hunting a ghost...

paich64, 2023-05-05:
Just an additional information :  The cartridge works perfectly when using the Ultimate Cartridge.

MJoergen, 2023-05-05:
After the game has run the first piece of code in the 1541 (mentioned above), the game then sends a larger chunk of code to the 1541 and runs it (starting at 0x0500), see below.

This appears to be some kind of custom loader.

```
0500   20 20 06   L0500     JSR L0620
0503   58                   CLI
0504   A9 00                LDA #$00
0506   8D 00 18             STA $1800
0509   A2 01                LDX #$01
050B   A0 08      L050B     LDY #$08
050D   AD 00 18   L050D     LDA $1800
0510   10 03                BPL L0515
0512   4C 82 05             JMP L0582
0515   29 05      L0515     AND #$05
0517   F0 F4                BEQ L050D
0519   4A                   LSR A
051A   A9 02                LDA #$02
051C   90 02                BCC L0520
051E   A9 08                LDA #$08
0520   7E 59 06   L0520     ROR $0659,X
0523   8D 00 18             STA $1800
0526   AD 00 18   L0526     LDA $1800
0529   29 05                AND #$05
052B   C9 05                CMP #$05
052D   F0 F7                BEQ L0526
052F   A9 00                LDA #$00
0531   8D 00 18             STA $1800
0534   88                   DEY
0535   D0 D6                BNE L050D
0537   78                   SEI
0538   CA                   DEX
0539   10 D0                BPL L050B
053B   A9 08                LDA #$08
053D   8D 00 18             STA $1800
0540   AE 56 06             LDX $0656
0543   AC 57 06             LDY $0657
0546   20 EA 05   L0546     JSR L05EA
0549   A0 02                LDY #$02
054B   B9 00 04   L054B     LDA $0400,Y
054E   29 83                AND #$83
0550   C9 82                CMP #$82
0552   D0 10                BNE L0564
0554   B9 03 04             LDA $0403,Y
0557   CD 59 06             CMP $0659
055A   D0 08                BNE L0564
055C   B9 04 04             LDA $0404,Y
055F   CD 5A 06             CMP $065A
0562   F0 2B                BEQ L058F
0564   98         L0564     TYA
0565   18                   CLC
0566   69 20                ADC #$20
0568   A8                   TAY
0569   90 E0                BCC L054B
056B   AC 01 04             LDY $0401
056E   AE 00 04             LDX $0400
0571   D0 D3                BNE L0546
0573   A2 02                LDX #$02
0575   8E 02 04   L0575     STX $0402
0578   A9 00                LDA #$00
057A   8D 00 04             STA $0400
057D   8D 01 04             STA $0401
0580   F0 1A                BEQ L059C
0582   A2 00      L0582     LDX #$00
0584   D0 08                BNE L058E
0586   A9 1A                LDA #$1A
0588   8D 02 18             STA $1802
058B   4C 05 D0             JMP $D005
058E   60         L058E     RTS
058F   C8         L058F     INY
[... 78 lines cut ...]
0639   4A                   LSR A
063A   4A                   LSR A
063B   A8                   TAY
063C   B9 46 06             LDA $0646,Y
063F   9D 00 07             STA $0700,X
0642   E8                   INX
0643   D0 F1                BNE L0636
0645   60                   RTS
0646   0F                   ???                ;00001111
0647   07                   ???                ;00000111
0648   0D 05 0B             ORA $0B05
064B   03                   ???                ;00000011
064C   09 01                ORA #$01
064E   0E 06 0C             ASL $0C06
0651   04                   ???                ;00000100
0652   0A                   ASL A
0653   02                   ???                ;00000010
0654   08                   PHP
0655   00                   BRK
0656   12                   ???                ;00010010
0657   01 03                ORA ($03,X)
0659   20 06 5B             JSR $5B06
065C   57                   ???                ;01010111 'W'
065D   2D 4D 05             AND $054D
```

MJoergen, 2023-05-05:
I believe I've found the cause of the infinite loop in hardware.

When the call to $FFB1 (LISTN) returns, the following happens (output from VICE):
```
10570814 .2153 075 024   16535445  A5 90       LDA $90      3e3600f2 06 01                                                         
10570815 .2155 075 027   16535448  C9 C0       CMP #$C0     003600f2 06 01
10570816 .2157 075 029   16535450  F0 17       BEQ $2170    003600f2 06 01
10570817 .2159 075 031   16535452  A9 6F       LDA #$6F     003600f2 06 01
10570818 .215B 075 033   16535454  20 93 FF    JSR $FF93    6f3600f2 06 01
```

In other words, the game reads from $90 and compares for equality with the value 0xc0. From [https://sta.c64.org/cbm64mem.html](https://sta.c64.org/cbm64mem.html) I find:
* Bit 6: 1 = End of file has been reached.
* Bit 7: 1 = Device is not present.

However, in hardware, the value 0x80 is read, i.e. only bit 7 is set.

So this seems to be a bug in the game, i.e. incorrect error handling. The game does not recognize that there is no disk drive present, and - even in hardware - continues through to $2159.

closed 2023-05-05 by sy2002