MJoergen/C64MEGA65 #26: Incorrect drive menu item selection when using simulated cartridges

MJoergen, 2023-04-28:
### Steps to reproduce:
* Enter OSM and mount a d64 image to drive 8 => The menu item is now selected and shows the file name.
* Enable "Simulate cartridge".
* Mount a CRT file => The CRT file now autostarts.

### Observed behavior
In the OSM the line for drive 8 now shows:
"=8: Mount Drive"

In other words, the file name is gone, the drive is no longer mounted, but the menu item is still selected.

If I then mount again a d64 image, I then get:
" 8: filename"

In other words, the drive is now mounted, but the menu item is no longer selected.

### Expected behavior
Consistency between drive menu selection and file name, i.e. both present or none present.

### Version
This is seen on WIP-V5-A17

sy2002, 2023-04-28:
@MJoergen Thank you, Michael. I will look into this. This is clearly a bug in the Shell:
Since the simulated cartridge is resetting the system and since resetting the system unmounts all drives, the "=" at drive 8 should be gone.

MJoergen, 2023-05-07:
This could be related to issue #33.

sy2002, 2023-05-13:
Fixed with commit https://github.com/MJoergen/C64MEGA65/commit/195b7cd2b88559e66bdb724589d8aee91e7df066

closed 2023-05-13 by sy2002