MJoergen/C64MEGA65 #27: SIMCRT: Vegetable deluxe crashes on start

sy2002, 2023-04-29:
In the file "Commercial_Releases.7z" that I've send you via Skype on 4/29/23, you will find the file `vegetables-deluxe-1.0.crt`. Here is AmokPhaze's test result (quote): "Vegetable deluxe won't work (blue screen and short / long reset won't help)"

sy2002, 2023-05-05:
@paich64 Can you confirm that this works on MiSTer and/or a real C64?

paich64, 2023-05-05:
I confirm Vegetables Deluxe works on Mister, with or without the commits you did in the Main Mister C64 core repository.

MJoergen, 2023-05-05:
I verified this works in VICE.

The cartridge type is 8 "Super Games", and is supported by MiSTer (as observed above).

From [https://codebase64.org/doku.php?id=base:crt_file_format](https://codebase64.org/doku.php?id=base:crt_file_format) the following is stated:
"The SUPER GAMES cartridge uses 4 16Kb banks ($8000-$BFFF) of ROM memory. Bank selecting is done by writing to $DF00."

MJoergen, 2023-05-05:
This cartridge copies the entire contents into RAM (starting at $0800), and then de-asserts GAME and EXROM (thus returning to "normal" mode) and jumps to $3800.

MJoergen, 2023-05-05:
Shortly after, it jumps to $C600, where it reads zero. It should instead read $78 (the SEI instruction).

MJoergen, 2023-05-05:
In VICE, the value $78 is read from address $BF00 when bank 2 is selected.

The problem seems to be the interaction between CORE/vhdl/crt_cacher.vhd and CORE/C64_MiSTerMEGA65/rtl/cartridge.v.

This particular game contains 4 chips, each of size 16 kB. The chips are numbered 0-3. However, MiSTer does a "shift left one bit" on the bank numbers, see these lines:
```
bank_lo <= {data_in[1:0],1'd0};
bank_hi <= {data_in[1:0],1'd1};
```
I'm not sure how to fix this, unless we completely replace MiSTer's cartridge.v.

paich64, 2023-05-05:
It's not going to help, but i now remember that "Vegetables" cartridge was working with V5A15 but "Vegetables deluxe" did not work with V5A15. And i tried again and yes whatever the core version is, "Vegetable deluxe" CRT won't run.

paich64, 2023-05-05:
This type of cartridge is surprisingly very marginal .....
In my full collection of more than 3000 CRTs, there are only 3 cartridges of this type  

Super_Games_(Colossus_Chess,_International_Football_&_Silicon_Syborgs).crt
falborg_c64_early_goldies.crt
Vegetables_Deluxe.crt    

They all fail in the exact same way.

MJoergen, 2023-05-05:
Thank you for this additional info. I'm considering re-implementing MiSTer's cartridge.v file, but have not decided yet.

MJoergen, 2023-05-07:
This has been fixed in commit #06bf4f0

closed 2023-05-07 by MJoergen