MJoergen/C64MEGA65 #32: have the configuration file remembering which .crt / .d64 was loaded and when the core starts have them loaded/mounted

paich64, 2023-05-06:
In the same way we have the C64 core starting a hardware cartridge when it's inserted or the 1541 drive mounting a floppy when it's powered on with a floppy inserted, 
Could we have the configuration file "remembering" which CRT / .D64 ?
If a .CRT was inserted last time the configuration was saved, then the core would start the .CRT immediately.
if a .D64 was mounted last time the configuration was saved, it would just be mounted by default when the core starts.
This way we can mimic something which could perfectly exist with true hardware.
Thanks !

sy2002, 2023-05-07:
Thank you for the feature request. I understand the use-case and I find it useful: I want to add this feature in future. This is something that needs a bit more work in the framework and therefore it will probably not make it in V5 of the core, but probably in V6.

nobruinfo, 2026-01-15:
@MJoergen @sy2002 Let me add a use case where this really is a showstopper. If you know the GEOS cartridges you also know they load lightning fast. No chance to HELP, RETURN, Cursor down, Cursor down, Cursor down... until the .D64 before GEOS halts apruptly because no disk drive is found. Same applies for those .crt that provide file browsers.
Possible workarounds: Either use a real cartridge or a real IEC attached drive. Like so the C64 core is a blast. 🥰 Thank you both for thousands of work hours spent into this adorable project!

sy2002, 2026-01-16:
Thank you for your kind words @nobruinfo . Yes - I understand the use case. Makes perfect sense. Unfortunately right now, neither @MJoergen nor me find time to improve the core. For the year 2026, I guess everyone "has to live with" core version 5.2 .

nobruinfo, 2026-01-16:
@sy2002 I had absolutely no intentions to put a date on your work. Take it as some sort of a prio wish should one of you find time anywhen in the future. It seemed important for me to descibe because I also know setups where resetting towards a nothing-mounted-situation of help.

sy2002, 2026-09-13:
Out of scope for V6.

nobruinfo, 2026-09-13:
@sy2002 No problem. All the existing features are a blast. The now in test V6 brings so many other things.

sy2002, 2026-10-09:
We can do this with a C64-specific M2M enhancement. Improves quality of life greatly. Added to V6 scope.