MJoergen/C64MEGA65 #34: Feature request: Remove OSM when auto-starting a PRG file or CRT file

MJoergen, 2023-05-07:
### Request
When a CRT file or PRG file is selected in the OSM, the game starts automatically. I would like the OSM display to disappear at the same time.

### Rationale
Often there are some nice graphics / splash screen / logo in the beginning, and this is partly obscured by the OSM.

sy2002, 2023-05-07:
* This is an interesting one. I totally get the rationale.
* I for example am more the type who fine-tunes the C64 setup for the game after I started it: Which SID? Joyport Switch?, etc.
* A solution that would possibly satisfy both of our needs would be to make it configurable.

sy2002, 2023-05-07:
Also related to this one (around Disk Mounting): https://github.com/MJoergen/C64MEGA65/issues/18

sy2002, 2026-09-13:
Out of scope for V6