MJoergen/C64MEGA65 #35: Bug in OSM: Confusion about what menu item is currently selected.

MJoergen, 2023-05-07:
### Steps to reproduce
1. In the OSM select a PRG file. The game autostarts. and in the OSM it shows `=PRG:shades.prg".
2. While this game is running, select a CRT file.

### Observation
Both the PRG file menu item and the CRT file menu item are selected.

### Expected behaviour
I would like to see in the OSM what game is currently running. So when a CRT file has been selected and autostarted, the PRG file menu item should no longer be selected. And vice versa.

sy2002, 2023-05-07:
Agreed - this is a bug.

sy2002, 2023-05-13:
Fixed with commit https://github.com/MJoergen/C64MEGA65/commit/3821d10aa4017943b1c49bf6c5c95db393d6f6b3

For clarity: This is a unidirectional thing: If a PRG was loaded and then afterwards a CRT then it is prudent to treat the PRG as "unloaded". The other way round is not true: If a CRT is loaded, for example a final cartridge III, which then boots into basic and you load a PRG afterwards, then the CRT stays "on"

closed 2023-05-13 by sy2002