MJoergen/C64MEGA65 #36: Cartridge only works if you load a certain other cartridge before

sy2002, 2023-05-07:
**Core is the newest Alpha 20**

Outrage and Metal Warrior Ultra are two modern high quality and high profile C64 games. I will send you the CRTs privately via Skype. Here is the bug-report that I received from Discord User KiDra#7570 via DM:

❌ Outrage.crt does not work. It starts with a black screen. But if you follow this sequence, it does work:

1. Turn the MEGA off
2. Load the Alpha 20 Core
3. Load Metal Warrior Ultra (filename in Skype: mwultra.crt): ✅ WORKS
4. Reset the core with a long press on the reset button. 
5. Load Outrage.crt: ✅  WORKS

KiDraDragon, 2023-05-07:
Here a little more info and a correction. The affected crt-image is not an EasyFlash format, but a generic one just named _Outrage.crt_. Both games come with multiple cartridge image types.

**MWUltra:**
* mwultra_easyflash.crt (works out of the box)
* mwultra_gmod2.crt (works out of the box)

**Outrage:**
* Outrage.crt (works only after loading _mwultra_easyflash.crt_ and resetting the core)
* Outrage_EasyFlash.crt (works out of the box)
* Outrage_GMod2.crt (works only after loading _mwultra_easyflash.crt_ and resetting the core)

Both MWUltra CRTs and _Outrage_EasyFlash.crt_ work out of the box.

_Outrage.crt_ and _Outrage_GMod2.crt_ only work when _mwultra_easyflash.crt_ was loaded first and the C64 core reset with a long press on the reset key. Then they work until the MEGA65 is power cycled.

Loading _Outrage_EasyFlash.crt_ does **not** cause the other versions to work correctly and neither does the GMod2-version of MWUltra. Only the EasyFlash-version of MWUltra has that effect.

MJoergen, 2023-05-08:
I've reproduced the bug.

The game Outrage works fine in VICE. The problem seems to be the game is reading from $DE00, which is not implemented here nor in MiSTer.

Does the file Outrage_GMod2.crt work on MiSTer ?

sy2002, 2023-05-08:
Downgraded priority from Prio 1 to Prio 5 because:
* Does not work with MiSTer either
* There is a workaround: Use the EasyFlash version of the `*.crt`
"Nice to have"

MJoergen, 2023-05-08:
> Still interesting, that they start working when the other cartridge was loaded first (that trick does not work on MiSTer).

Indeed. It's more of a bug than a feature (but who is to tell the difference!)

GMod2 cartridges have an on-board EEPROM, and the busy bit is read as bit 7 of $DE00. In our case we're essentially reading an uninitialized variable.

It's described here: http://wiki.icomp.de/wiki/GMod2

As far as I can tell, in VICE the EEPROM implementation is in src/core/m93c86.

MJoergen, 2023-05-09:
The datasheet for the EEPROM installed on the GMod2 cartridge can be found [here](https://www.st.com/content/ccc/resource/technical/document/datasheet/00/a9/de/01/11/0d/4d/b1/CD00001142.pdf/files/CD00001142.pdf/jcr:content/translations/en.CD00001142.pdf).

As far as I can tell, it basically says that the serial data pin from the EEPROM is used to indicate ready. This agrees with the program Outrage, where it at some point waits for the serial line to go high.

This can be hacked a number of ways. Probably the cleanest is just to return 0x80 (or 0xFF) when a GMod2 cartridge is reading from $DE00.

MJoergen, 2023-05-09:
Fixed in commit d4a4456

paich64, 2023-05-09:
> I just tested Outrage_GMod2.crt (and Outrage.crt) on MiSTer, neither works. I guess it is a non-issue, because Outrage_EasyFlash.crt works (on C64MEGA65 and MiSTer).
> 
> Still interesting, that they start working when the other cartridge was loaded first (that trick does not work on MiSTer).

It's not a non-issue as Outrage is a commercial game still being sold on protovision / psytronic and as far as i know The EasyFlash Release is not delivered by both of them :)

closed 2023-05-10 by sy2002