MJoergen/C64MEGA65 #40: E2IRA demo: Works on V4 and on MiSTer but crashes on V5 Alpha

sy2002, 2023-05-10:
E2IRA demo: Works on V4 and on MiSTer but crashes on V5 Alpha

paich64, 2023-05-21:
I have investigated further and curiously (but i don't explain why since i'm 100% sure i have already run it on my C64 + ultimate cartridge), It also fails (in the exact same way as on V5) on my C64 with the Ultimate cartridge, but runs fine on the Ultimate 64.
If i manage to understand what would differ between the Ultimate cartrdige firmware and the Ultimate 64 firmware maybe i can have a beginning of explaination.

To be as clear as possible:
- It was running fine on V4
- It fails on V5
- It runs fine on Mister
- It runs fine on Ultimate 64
- it fails on real C64 with Ultimate cartridge.

sy2002, 2023-05-21:
@paich64 Let's focus on MiSTer: What would help us to spot the code changes more easily: Can you please test if it runs fine on **these three older MiSTer versions**:

*  Release 20221007
*  Release 20221124
*  Release 20230224

paich64, 2023-05-21:
Actually focusing on ultimate cartridge helped : I asked on Ultimate Facebook group if someone was aware of the issue with ultimate cartridge. They pointed me to this file [https://csdb.dk/getinternalfile.php/229680/e2ira_101_by_arise.zip](https://csdb.dk/release/download.php?id=269249)
There was a known timing issue with Ultimate Cartridge and the group re-released it as e2ira_101_by_arise.zip. Specifically for Ultimate cartridge, but guess what ? It also fixed the issue on our V5 core. So do you still want to move forward ?

paich64, 2023-05-21:
I understand it can make sense to investigate further with the Mister Cores. I have no problem with this. Just let me know :)

paich64, 2023-05-21:
By the way if you check the discussion here : [https://csdb.dk/release/?id=218343] you will see that the original demo was also failing on few real 1541 drives, hence the re-release.

sy2002, 2023-05-23:
@paich64 Update: I did the code comparison (our source code and MiSTer's) and literally did not find ANYTHING that could explain why it works with V4 (and with MiSTer) and why it does not with V5 other than our slow-down mechanism due to **"HDMI: Flicker-free"** which we changed in V5. Can you please re-test the demo on V5 while "HDMI: Flicker-free" is **switched-off**?

paich64, 2023-05-23:
@sy2002 I tested with "HDMI: Flicker-free" switched off and it does not help unfortunately. Tomorrow i will test the demo on the different mister core and report here about the results.

sy2002, 2023-05-25:
@paich64 I re-read the comment from "wacek" the author of the demo, who also created the fix: "Somehow the UII+ emulates slower disk speed than UII, Vice etc."

We changed the clock frequency relationship between the C64 core and the simulated 1541 drive. This is in line with the symptom - and I guess we are talking about a very tight clock cycle here.

We cannot change this back on our side to like it was on V4 without breaking other things.

So since the bugfixed version of the demo works with V5, I will close this one as "wontfix".

closed 2023-05-25 by sy2002