MJoergen/C64MEGA65 #43: HDMI-only: Black screen after power-on or bitstream loading

sy2002, 2023-05-10:
Black screen from time to time on startup (independent if a cartridge is inserted or not): HDMI-only: VGA shows a normal C64 start screen including a blinking cursor. Problem:

* On HDMI the screen stays black
* But you can use the OSM via the Help key (??!)

MJoergen, 2023-05-23:
@sy2002 : Do you have statistics on this bug, i.e. how often it happens ?

Has this been seen when loading image via JTAG ? Or is it only after power-on ?

sy2002, 2023-05-23:
@MJoergen I have seen this via JTAG and via power-on but it happens infrequently / seldomly.

In contrast to that, issue https://github.com/MJoergen/C64MEGA65/issues/53 happens 100% of the time.

sy2002, 2023-06-04:
Still happens, but happens so seldomly that we won't fix this for V5 and let's see if it's still worth an own issue in the future. Closing as wontfix.

closed 2023-06-04 by sy2002