MJoergen/C64MEGA65 #53: Black screen on startup when in MONITOR mode

sy2002, 2023-05-10:
When we start the core in debug mode (MONITOR operating system instead of Shell firmware) then the VGA and HDMI stays black (maybe the core stays in reset?).

Expected is that we would see a normal C64 screen (without ASCAL filters and without help menu but a normal C64 screen).

This worked with V4 and maybe some of the hardware cartridge changes introduced this regression and MAYBE this regression is the root of some more obscure reset/startup problems

sy2002, 2023-05-23:
@MJoergen I just saw that you assigned this one to yourself. Thank you for supporting! :-)

Here is my hypotheses about this bug. Might be completely wrong, but maybe a starting point for your investigation:

```vhdl
   -- Ensure that the cpu_data_in process provides a potential cartridge's ROM to the CPU so that we can
   -- start a cartridge directly upon power on (aka "cold start"). The complex reset mechanisms in the
   -- system create two hard_reset_n signals directly after power on, so we need to compensate for that.
   handle_cold_start : process(clk_main_i)
   begin
      if rising_edge(clk_main_i) then
         if hard_reset_n = '0' and hard_rst_counter = C_HARD_RST_DELAY and (system_cold_start = 4 or system_cold_start = 2) then
            system_cold_start <= system_cold_start - 1;
         elsif hard_reset_n = '1' and hard_rst_counter = 0 and (system_cold_start = 3  or system_cold_start = 1) then
            system_cold_start <= system_cold_start - 1;
         end if;
      end if;
   end process;
```
(see also https://github.com/MJoergen/C64MEGA65/blob/develop/CORE/vhdl/main.vhd#L470)

If this is the reason (I did not check yet), then the solution might be to use different counters in case of QNICE_FIRMWARE equals QNICE_FIRMWARE_M2M. Some code generator that says:

if QNICE_FIRMWARE = QNICE_FIRMWARE_M2M then your reset constants are x, y, z else a, b, c  and then replace the magic numbers in the code by constants.

Context: The Shell does certain "reset things" depending on how config.vhd is being setup. It holds the core in reset for example to load the saved OSM settings from the config file. When the Shell is not loaded, this code segment might "wait for one reset too much" - which is not happening due to the Shell being loaded.

MJoergen, 2023-05-24:
Fixed in commit 92b91ed

sy2002, 2023-05-25:
Tested it 10 times in a row: Fix works. Thank you. I am closing this.

closed 2023-05-25 by sy2002