MJoergen/C64MEGA65 #55: HyperRAM/TreuLove issues

sy2002, 2023-05-10:
* TreuLove is not working for @mpryon and @muse on V5A10:
  https://discord.com/channels/@me/1034779919802191882/1087995656163041291

* Even though he seems to be on a Rev D, @mpryon does not have any glitches
  with TreuLove any more since V5A9:
  https://discord.com/channels/@me/1034779919802191882/1087336036977344532
  https://discord.com/channels/@me/1034779919802191882/1087445661827469442

* But @muse still has glitches with V5A9
  https://discord.com/channels/@me/1034779919802191882/1088000630611791893

* HyperRAM Rev D: @muse and @mpryon having CRASHING TreuLove when using the
  Alpha 10 core versus the Alpha 9 core (?!). Alpha 10 actually did not
  change anything significant versus Alpha 9. We ruled out simple reasons
  such as "wrong config file":
  https://discord.com/channels/@me/1034779919802191882/1089138523912802344
  https://discord.com/channels/@me/1034779919802191882/1089989286092361838

Since these discors links are only readable for me (sy2002): I need to first describe the situation better here in GitHub before we can tackle it.

sy2002, 2023-06-17:
Discord user "muse" aka on GitHub as @sho3string confirmed on June 10, 2023 (Skype conversation) that this issue is still reproducible with our Version 5 RELEASE CANDIDATE (WIP-V5-B4).

sy2002, 2023-08-29:
Maybe related to this one: https://github.com/MJoergen/C64MEGA65/issues/86

sy2002, 2023-11-02:
Quotes from a Skype conversation between MJoergen and sy2002:

MJoergen:

I have an idea to tackle GitHub issue 55. My hypothesis is to change line 292 in digital.vhd from "N_BURST   => 256" to "N_BURST   => 128". This will reduce the maximum time the C64 core needs to wait before getting access to the HyperRAM.

I'm fully aware of your vigilance regarding graphics. So there are two questions to answer:
1. Does this change actually fix the problem seen (i.e. the specific issue)?
2. Does this change have any negative side-effects (e.g. graphics glitches or other problems)?

I'm 99.999999% sure that 2) is not a problem. However, this is something that is best verified in practice, probably with the help of Olivier. So such a change - if implemented - should be carefully planned.

sy2002, 2023-11-02:
@sho3string (aka muse) and @mpryon can you please re-test the TreuLove issue with this Alpha 3 version of the upcoming V5.1 core of the C64 (attached to this comment)? @MJoergen came up with an idea that might fix the bug.
[c64v51a3.zip](https://github.com/MJoergen/C64MEGA65/files/13244399/c64v51a3.zip)

sho3string, 2023-11-03:
I tested two versions ( reu1750 release and the first release )

No glitching in both cases. Well done guys.

sy2002, 2023-11-03:
Thank you @sho3string ! :-)

closed 2023-11-03 by sy2002