MJoergen/C64MEGA65 #64: SIMCRT: V5A23 - GMOD2 TESTS - Hunters moon remastered gfx glitch

paich64, 2023-05-11:
hunters moon remastered (GMOD2) shows a GFX glitch when starting a game.
The main little popup where it's normally written that you have to select if you want to start a new game or contine a game is screwed up (we have blank horizontal bars instead of having the text).

Otherwise the game is playable.

On Mister core (which has the latest commit brought by Michael), it's even worse : The game simply freezes before having the popup.

sy2002, 2023-05-11:
Thank you @paich64 for submitting this bug. @MJoergen I suggest to move this to V6 (or later) because the bug is also on MiSTer.

paich64, 2023-05-11:
[image]
[image]

paich64, 2023-05-11:
The correct screenshot is from vice