MJoergen/C64MEGA65 #66: SIMCRT: V5A23 - OCEAN TYPE TESTS - a few cartridges get a  hanging blue screen after having been loaded

paich64, 2023-05-11:
The following OCEAN TYPE 1 cartridges get a hanging blue screen after having been loaded :

Leader Board - Pro Golf Simulator.crt
Wonderboy.crt
tennis_ccs.crt
Magic Desk I (1983)(Commodore).crt
Contiki Operating System, The (200x)(-)[cr HF].crt

Behaviour on Latest Mister core :

Leader Board - Pro Golf Simulator.crt                   : Starts after the blue screen (the borders quickly flash then the game starts)
Wonderboy.crt                                                       : Starts after the blue screen (the borders quickly flash then the game starts)
tennis_ccs.crt                                                          : Does not start after blue screen, the borders keep flashing slowly 
Magic Desk I (1983)(Commodore).crt                    : does not start after blue screen
Contiki Operating System, The (200x)(-)[cr HF].crt : does not start after blue screen

MJoergen, 2023-05-14:
"Leader Board - Pro Golf Simulator.crt" : This file seems corrupt. Although it does seem to work in VICE, during startup it seems to execute garbled code. Presumably because during initialization the Kernal tries to execute a function in BASIC, but this is banked out.

Here is the relevant part of the debug output from VICE:
```
.E422 272 047    1648631  A5 2B       LDA $2B      000000fd 1f 00
.E424 272 050    1648634  A4 2C       LDY $2C      010000fd 1f 00
.E426 272 053    1648637  20 08 A4    JSR $A408    010008fd 1f 00
.A408 272 059    1648643  FF FF FF    ISB $FFFF,X  010008fb 1f 00
.A40B 273 003    1648650  FF FF FF    ISB $FFFF,X  010008fb 1f 00
```

MJoergen, 2023-05-14:
Exactly the same happens in Wonderboy.crt

paich64, 2023-05-14:
Something i have noticed with leaderboard is that the core does not seem to be hung : If you check carefully it looks like the screen is switched from 40 columns mode to 38 columns mode then to 40 columns mode then to 38 columns mode and so on ... and never stops doing this. This does not tell us how it comes that on mister this does not seem to be a problem.

sy2002, 2023-05-20:
@MJoergen What will we do with this?  @paich64 confirms it works with MiSTer - which is strange.

If you decide to do "Close as not planned" then please also change the label to invalid.

MJoergen, 2023-05-22:
I'm closing this issue, since the files seem corrupt. The fact they work on the MiSTer seem more like a coincidence.

closed 2023-05-22 by MJoergen