MJoergen/C64MEGA65 #67: SIMCRT: V5A23 - OCEAN TYPE TESTS - Cyberball.crt gets a black screen after the main menu

paich64, 2023-05-11:
Cyberball.crt loads successfully, but after we reach the main screen, hitting fire to start the game leads to a black screen. The game won't start

Behaviour on the latest Mister core :

Nearly similar : after we reach the main screen, the screen gets black with a few corrupted characters. The game won't start

sy2002, 2023-05-11:
Thank you @paich64 for reporting. @MJoergen I suggest to move this to V6 or later since it does not work with MiSTer, too.

MJoergen, 2024-11-11:
When I try with VICE version 3.8 I also get the black screen when starting the game.

@paich64 Can you get this game to work on VICE ?

paich64, 2024-11-12:
> When I try with VICE version 3.8 I also get the black screen when starting the game.
> 
> @paich64 Can you get this game to work on VICE ?

Yes i can make it work with vice 3.8 : https://csdb.dk/release/download.php?id=292291

Maybe attempt this :  
1) save your current settings using "preferences-> save settings to...."  (and choose a name like "non_working_conf.ini")
2) reset vice settings to default using "preferences-> restore default settings" 
3) save again your current settings using "preferences-> save settings to...."  (and choose a name like "default_conf.ini")

Test if the .crt works (first load settings from "default_conf.ini").
If it works they comparing both setting files might give a clue.

paich64, 2024-11-12:
> > When I try with VICE version 3.8 I also get the black screen when starting the game.
> > @paich64 Can you get this game to work on VICE ?
> 
> Yes i can make it work with vice 3.8 : https://csdb.dk/release/download.php?id=292291
> 
> Maybe attempt this :
> 
> 1. save your current settings using "preferences-> save settings to...."  (and choose a name like "non_working_conf.ini")
> 2. reset vice settings to default using "preferences-> restore default settings"
> 3. save again your current settings using "preferences-> save settings to...."  (and choose a name like "default_conf.ini")
> 
> Test if the .crt works (first load settings from "default_conf.ini"). If it works they comparing both setting files might give a clue. ![cyberball](https://private-user-images.githubusercontent.com/6482872/385189871-b045bff7-4fe5-4445-ae43-6c9ea299699f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzOTg0NTUsIm5iZiI6MTczMTM5ODE1NSwicGF0aCI6Ii82NDgyODcyLzM4NTE4OTg3MS1iMDQ1YmZmNy00ZmU1LTQ0NDUtYWU0My02YzllYTI5OTY5OWYuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTExMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMTJUMDc1NTU1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzI4NTlhZjBhYTEwZDRlNjRkNTU3YWYzNjRlN2NiZWQ3YTliMjBmZjQ5MGMzNzg1NzhlZjA1Yzk5YWZhY2RmMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.o_YkDTR1LhVZZkSSWm7e7bMAUwhX5dMdlwunIVPkJTY)

Ah wait it's when starting the game, let me check again.

paich64, 2024-11-14:
> Yes i can make it work with vice 3.8 : https://csdb.dk/release/download.php?id=292291
>
> Maybe attempt this :
>
> 1. save your current settings using "preferences-> save settings to...."  (and choose a [...]

Indeed it won't start in vide 3.8..... however, i think i understand what's wrong : 

1) When the game start you can see it's a "Domark" game, and i was surprised to find it tagged as an ocean type cartridge as "Domark" is known to have its on cartridge format 

2) And indeed, if you check vice documentation, you can see that cyberball is referenced as MAGIC DESK cartridge and not an Ocean type cartridge : 
[image]

3) If I open the cartridge file with a binary editor i can see it's tagged as an Ocean Type cartridge (5) : 
[image]

This cartridge is tagged as OCEAN TYPE (5) while it's supposed to be of type 19 ($13)

4) So i decided to modify the cartridge type :  
[image]

5) I Launched the modified .crt and it works in Vice 3.8 !  
[Cyberball_modified.zip](https://github.com/user-attachments/files/17757643/Cyberball_modified.zip)

6) I launched the modified .crt with core V5.1 and now i can play the game.

So obviously here we have an incorrect release (but widely spread over the web). Modifying the cartridge type in the .crt file is enought to make the game working.

I guess the issue can be closed :)

sy2002, 2024-11-14:
Closing as @paich64 suggested.

closed 2024-11-14 by sy2002