MJoergen/C64MEGA65 #70: SIMCRT: FinalCartridge Freezer Key question

sy2002, 2023-05-20:
This is not a bug request but more a question / double-check:

Is it intentional, that in [cartridge.vhd](https://github.com/MJoergen/C64MEGA65/blob/develop/CORE/vhdl/cartridge.vhd#L199) in the Final Cart III section the input signal `freeze_key_i` is being used directly, while there are in parallel signals like

```vhdl
   freeze_req <= not old_freeze and freeze_key_i;
   freeze_ack <= nmi_o and not old_nmiack and nmi_ack_i;
   freeze_crt <= freeze_ack and not mod_key_i;
```

MJoergen, 2023-05-22:
Well, the "easy" answer is that this is what MiSTer does :-)

That being said, there might be a bug (common to our core and the MiSTer core) that should be fixed.

So, I'm changing this to "Bug" for the time being, until I've investigated this some more.

MJoergen, 2023-05-22:
Fixed in commit 0be837f

closed 2023-05-22 by MJoergen