MJoergen/C64MEGA65 #71: SIMCRT: Final Cartridge's Monitor does not properly return

sy2002, 2023-05-20:
I am not fully sure if we want or even can fix this one because I am experiencing the same behavior when I use a real Final Cartridge III hardware cartridge so this might be a C64 MiSTer core bug and not a specific cartridge.vhd bug:

1. Leave the FCIII Desktop and go to Basic
2. Enter a program like ```10 PRINT "HALLO": GOTO 10``` and run it
3. Press the freezer key <kbd>Restore</kbd> (due to question https://github.com/MJoergen/C64MEGA65/issues/70 you might need to hammer multiple times on the key)
4. Go to the Monitor (menu EXIT/Monitor)
5. Press <kbd>x</kbd>
6a. Behavior: You land at the `READY.` prompt
6b. Expected behavior: The program you entered in (2) is continuing where it left off

In general, the FCIII simulation is able to do this "continue where you left off": If you perform the above mentioned reproduction steps but do NOT enter the Monitor but choose EXIT/RUN instead, then the BASIC program will nicely continue where it left off.

Feel free to add the label "V6 or later" if you feel this is out of scope due to the hardware version of the FCIII also not working with our core, yet.

MJoergen, 2023-05-22:
I'm setting this to V6 for now, since I consider this a "nice-to-have" and there is a work-around.

Kugelblitz360, 2024-07-31:
This is not a bug at all, this is expected behaviour. FCIII does not have RAM. When you enter the freezer, a LOT of dirty tricks are used to show a menu without using any bitmap RAM (this is all ROM-based sprites basically). When you enter the monitor, FCIII really needs some RAM and after that you can not return. Documented by @dmantione in the Forum64 threads.

sy2002, 2024-08-03:
Thank you @Kugelblitz360 - this is very helpful info. I am closing this one as WONTFIX.

closed 2024-08-03 by sy2002