MJoergen/C64MEGA65 #76: Sprites not shown correctly

sy2002, 2023-06-09:
User @Schwefelholz reported the following issue (https://github.com/sy2002/MiSTer2MEGA65/issues/17):

The game "Raid on Bungeling Bay" that shows a "ghost image" of the helicopter's rotor Blade here is how it looks:

[image]

And here is how it looks on a real C64 and/or on an Ultimate 64:

[image]

@paich64 Can you please double-check / confirm if this also happens on the MiSTer?

paich64, 2023-06-10:
Just tested the game on MEGA65 c64 core V5 beta3 : No such issue .....
[image]

Schwefelholz, 2023-06-10:
Hmm, that's strange. I had this issue on MEGA65 c64 core V5 beta3 as well.

Maybe just another copy of the game? I'll try once again and will report back.

sy2002, 2023-06-10:
@Schwefelholz This version is known to work without glitches:
[Raid_on_Bungeling_Bay.zip](https://github.com/MJoergen/C64MEGA65/files/11712031/Raid_on_Bungeling_Bay.zip)
I am closing this issue.

closed 2023-06-10 by sy2002