MJoergen/C64MEGA65 #79: Long reset does not reset "Eagle's Nest"

sy2002, 2023-06-19:
This bug is a regression introduced in V5 as this worked in V4.

Expected behavior (see also tests/README.md):

* Short reset: Back to game's title screen
* Long reset: Reset C64, back to Basic

What happens: A long reset does not return to Basic

sy2002, 2023-06-22:
Confirmed by discord user mauricevg that Pac-Man is affected, too:
https://discord.com/channels/719326990221574164/794775503818588200/1121155136853639219

sy2002, 2023-10-29:
Fixed.

closed 2023-10-29 by sy2002

MauriceMEGA65, 2023-11-20:
I just discovered: Pac-man needed to have the hard reset to be fixed, which is done in one of the alpha versions.
However, I just encountered the following:

Hard-reset DOES properly reset the system, I'm back in the BASIC 2 or JiffyDOS screen
When I do a soft-reset AFTER the hard-reset, I'm back in the Pac-man welcome screen!?!?

sy2002, 2023-11-21:
@MauriceMEGA65 I think this is "works as desgined". What happens on a real C64?

MauriceMEGA65, 2023-11-21:
A real C64 does not have a reset button IIRC. It’s no dealbreaker of course, just I thought the hard-reset is like a power cycle on a real C64. But apparently the memory is not really reset as a soft-reset after the hard-reset seems to reinitialize what was in memory before the hard-reset.

sy2002, 2023-11-21:
Yes, this is what I mean by "works as designed" :-) The hard reset does what a hard reset button (that was an extra gadget you could buy), did. We implemented this behavior: https://www.c64-wiki.com/wiki/Reset_Button

MauriceMEGA65, 2023-11-21:
Ah! “_Previously used data in the [RAM](https://www.c64-wiki.com/wiki/RAM), especially [BASIC](https://www.c64-wiki.com/wiki/BASIC)-programs, survive this_”
Yes, all clear now! Thank you and sorry for the confusion!