MJoergen/C64MEGA65 #82: Support MEGA65 R4

sy2002, 2023-07-03:
This enhancement is tightly coupled with this enhancement of the MEGA65 framework: https://github.com/sy2002/MiSTer2MEGA65/issues/19

Currently, there are a lot of open questions and for the time being (given the limited amount of time that we currently can invest into the C64 core) it might be valid to look at the MEGA65 R4 support of the C64 core as independent from the enhancements of the M2M framework: The open question is: Quick and dirty C64 or "do it right" in the M2M framework.

This is the first question that needs to be answered.

MJoergen, 2023-10-20:
The plan is to develop and test on this repo, and then later (or concurrently) merge in the changes to the M2M framework.

The following list is copied from https://github.com/sy2002/MiSTer2MEGA65/issues/19 :

- [ ] HDMI driver IC
- [ ] Removal of MAX10: Reset-detection
- [ ] Analog audio DAC with I2C config interface
- [ ] 64 MB SDRAM
- [ ] Bi-directional joystick ports: (How to) support? Paul describes a mouse problem on the MEGA65 with new port; mouse support for C64 => GEOS, etc. needs double-check
- [ ] New RTC chip (M2M does not support RTC, yet, but would be nice, e.g. for GEOS but also for tagging `*.d64` with the right date/time when the core writes to a disk image: We need to support 3 different RTCs as it looks like: Original M65 R3 RTC, "workaround R3 RTC" and new R4 RTC)
- [ ] IEC has a new and different circuit incl. one signal that is now negated

sy2002, 2023-10-20:
For our convenience: Here are the two links to the filtered lists of M2M issues and C64 issues that are currently **IN SCOPE** for V5.1 of the C64:

**C64 issue tracker**
https://github.com/MJoergen/C64MEGA65/issues?q=is%3Aopen+is%3Aissue+label%3A%225.1+(R4%2FR5+Surge)%22+

**M2M framework issue tracker**
https://github.com/sy2002/MiSTer2MEGA65/issues?q=is%3Aopen+is%3Aissue+label%3A%22R4%2FR5+Surge%22

MJoergen, 2023-10-21:
Re. HDMI: The physical HDMI connector has a pin called "+5V" where the source (i.e. the MEGA65) must supply power. On the R3 board this is done programmatically from the FPGA via the HDMI driver chip U10 = TPD12S016PWR. But on the R4 board the pin on the HDMI connector is hard-wired to the +5V power supply on the board. So the FPGA is not involved at all.

sy2002, 2023-11-02:
Closing this issue because our Alpha 3 version includes everything but SDRAM and RTC: Both will be after V5.1. We have separate GitHub issue for these two features.