MJoergen/C64MEGA65 #83: Demo "Royal Arte Final" fails to load Disk n°2 since V5 Alpha 14

paich64, 2023-07-03:
- Since V5 Alpha 14, Demo "Royal Arte Final" (https://csdb.dk/release/?id=11619) does not load 2nd disk.
- Issue still exists in V5Final
- To reproduce : 
Use any core V5 >= V5A14
Load disk 1
run
let the intro play, it will then load part 1
let part 1 run a few seconds (or more if you like) and press space to skip to part 2
let part 2 run a few seconds (or more if you like) and press space to skip to part 3
Let part 3 run a few seconds (or more if you like) and press space
"Booze Design" logo appears at the bottom of the screen, you can see some disk led activity
A text asking to swap to disk 2 and then press space appears
Mount disk 2, Then, press space
Nothing happens.

in V5A11, V5A12, after having pressed space the demo keeps running the parts from disk 2.

I do not have V5A13 core to test might have appeared in V5A13.

but for sure : it was still working in V5A12 and stopped working in V5A14
[2001 - Royal Arte.zip](https://github.com/MJoergen/C64MEGA65/files/11941094/2001.-.Royal.Arte.zip)

sy2002, 2023-07-04:
@paich64 Thank you. Tracking down the Alpha versions where it did work and where it stopped working helps. Here is a code comparison between Alpha 14 and Alpha 12:

https://github.com/MJoergen/C64MEGA65/compare/dcb2196..7711653

sy2002, 2024-01-13:
Fixed with V5.1 Alpha 7.

closed 2024-01-13 by sy2002