MJoergen/C64MEGA65 #84: Original "ZetaWing" cartridge stopped working with V5 Alpha 17

paich64, 2023-07-04:
Before V5A17 i could play with my original "ZetaWing" Cartridge.

Starting with Core V5A17 (which fixes the issue i had with EasyFlash 1), ZetaWing does not work anymore.
I simply get a black screen, then the screen is filled up with characters and nothing else happens. 

                               ; V5A10  ; V5A12 ; V5A14  ; V5A16  ; V5A17
Zetawing cartridge ; OK       ;  OK      ; OK        ; OK       ;   KO 
EasyFlash 1             ; N/A     ; N/A      ; KO        ; KO       ;   OK

paich64, 2024-01-13:
@sy2002 you can close this one, it's 100% related to the problem i have with the cartridge port on my R3A board.

sy2002, 2024-01-14:
OK @paich64 - I closed it.

closed 2024-01-14 by sy2002