MJoergen/C64MEGA65 #85: Audio level control

eliyo, 2023-08-18:
Unless I am mistaken I didn't see a volume control for the audio out, so the sound is too loud when using headsets where we cannot control the volume. Would be great to add this feature.

sy2002, 2023-08-21:
@eliyo Great point. Thank you for the feature request. We will add it to one of the next releases.

sy2002, 2023-10-20:
Step #1: Research and decide, if we want to do this for version 5.1:
* How does Audio volume control work at all on a fundamental level, given that humans hear volume logarithmically and the Decibel scale is logarithmic, too
* The R4/R5 analog audio DAC offers volume control
* R3/R3A (and R2) does not
* How do we do it for HDMI