MJoergen/C64MEGA65 #87: Add an option to change the Processor Clock Frequency

oe3gwu, 2023-09-26:
It would be nice to have the same clock options as on the MEGA65. For example 1, 2, 4 , ... 40 MHz.

The reason is that Test Drive II is utterly really slow using 1MHz. And it is kinda playable on a C128 using 2MHz and it rocks using 4MHz using my Ultimate 64.

sy2002, 2023-09-27:
@oe3gwu Nice feature request. Thank you. The underlying MiSTer core does support several turbo modes, see

https://github.com/MJoergen/C64_MiSTerMEGA65/blob/1c6bd485e7d71d4c89348a82ae875de739d7e267/c64.sv#L261

and

https://github.com/MJoergen/C64_MiSTerMEGA65/blob/1c6bd485e7d71d4c89348a82ae875de739d7e267/c64.sv#L937

so @MJoergen when / as soon as we tackle this one we might want to research how these work.

MJoergen, 2023-11-01:
Remember to take into account #2 so we avoid any regression bugs

sy2002, 2026-11-05:
### Update: what is coming

Here is where we stand and what you can expect. This issue now also covers #242; thank you @Kugelblitz360 for the detailed research and for the two patched games, which will be great test cases.

The turbo modes will come with **Alpha 22X1** of V6. They are the turbo modes of the MiSTer C64 core, nothing more and nothing less:

* You will find them in the **Model** submenu: **Turbo mode** `Off` / `C128` / `Smart` and **Turbo speed** `2x` / `3x` / `4x`.
* **Off** (default): a standard C64.
* **C128**: the software decides. Turbo is only active while a program switches it on via bit 0 of `$D030`, the 2 MHz register of the C128. Instead of the 2 MHz of the C128, the core then runs at the speed you selected. Ordinary C64 software never touches this register and runs at normal speed, so this mode is the right choice for software that supports it.
* **Smart**: always fast, but the core automatically falls back to normal speed for half a second after any activity on the IEC bus, so loading from the simulated drives and from real drives on the IEC port keeps working. Great for slow 3D games. Timing-critical games and demos will break, so switch turbo off for those.
* The menu shows at a glance when turbo is on, for example `Model: PAL Smart 4x`, and the setting is saved like every other setting.

What stays the same:

* The picture stays correct in turbo mode (unlike on a real C128 at 2 MHz), because the VIC-II keeps its own memory access slot. Badlines, sprites and the I/O chips (VIC-II, SID, CIAs) keep their normal C64 timing, so the real speed-up depends on the program and is somewhat below the nominal 2x, 3x or 4x.
* CIA timers, the time-of-day clock, HDMI flicker-free and NTSC are not affected, because turbo does not change any clock.
* REU transfers and the bank switching of simulated cartridges keep running at normal speed.
* Real cartridges in the expansion port keep working: every access to the expansion port runs at normal C64 speed, and the cartridge always sees a normal 1 MHz PHI2 clock. We prepared this already in June. While preparing the turbo work we found one more case in which a real cartridge could get too little time to answer, and we will fix it before the alpha is released.

What is not coming: speeds above 4x (such as the 40 MHz of the original request), the turbo registers of the C64 Ultimate (`$D031`) and of the Turbo Chameleon (`$D0F3`), SuperCPU registers and a "no badlines" mode. We deliberately stay with what the MiSTer core supports.

Testing: as soon as the alpha is ready, we will ask for testers here. We will especially need people who test games, such as Test Drive II (the game that started this issue), Uridium+ and Boris's version of Rescue on Fractalus, in all three modes and at all speeds. On our side, we will run the usual regression tests and test real cartridges, SIMREU (especially on R3) and the drift tests from #2.

sy2002, 2026-11-06:
@Kugelblitz360 @paich64

**Alpha 22X1** delivers Turbo Modes: [C64MEGA65-WIP-V6-A22X1.zip](https://github.com/user-attachments/files/34473032/C64MEGA65-WIP-V6-A22X1.zip)

Would be great if one or both of you could confirm a basic green light (like "in general works") and as an optional but valuable addition, add some more sophisticated tests:

* IEC devices
* Hardware cartridges
* Simulated REUs and cartridges
* PAL and NTSC subtleties
* Flicker-Free Modes

**Please document results here, not on Discord**

sy2002, 2026-11-10:
Got this feedback from @Kugelblitz360 via GitHub: "Doing some Turbo tests right now, so far works absolutely as planned, with the added bonus that $d030 does not blank the screen as on the C128."

Good enough as a feedback for me to close this issue. Bugs will be treated as separate issues.

closed 2026-11-10 by sy2002