MJoergen/C64MEGA65 #88: PRG loader fails at 4k intro Boo by Reflex

sy2002, 2023-09-27:
The following is true for our core as well as for the MiSTer core, tested by @paich64 (and for sy2002's internal records, here is the private Discord link: https://discord.com/channels/@me/1034779919802191882/1156677319390412880):

How to reproduce:

1. Download the PRG of Boo by Reflex, for example from here: https://csdb.dk/event/?id=3187
2. Use MiSer's or C64 for MEGA65's "direct PRG loader": Fails
3. Put the prg on a D64 disk image, load it manually: Works

sy2002, 2024-08-18:
Might be related to this bug: https://github.com/MJoergen/C64MEGA65/issues/144

csoren, 2024-12-11:
Boo uses the 1541 CPU. This issue seems to be caused by the drive not being mounted until an image is mounted. Loading the .PRG and then quickly mounting a .D64 will make the intro start up. I have also tried forcing the drive mounted in the HDL, which also makes selecting the .PRG work.

sy2002, 2025-02-22:
@csoren Thank you this is VERY valuable feedback. But then this is a slightly larger thing to fix for us, so I will remove the "V5.2" flag and put it into "V5.3"

paich64, 2026-09-02:
@sy2002 Are you sure you included the fix in V6A20X2 ? Because the .prg still fails loading.

paich64, 2026-09-02:
@sy2002 MY BAD !!!!!! I had forgotten to set Drive 8 to "always" in drive settings :) 

It's all good. With Drive 8 set as "always", the intro perfectly starts and works :) 

Issue fixed, you can close it :)

closed 2026-09-13 by sy2002