MJoergen/C64MEGA65 #90: Add Support for the internal 1565 Drive of the MEGA65 R3

oe3gwu, 2023-10-14:
It would be nice to have the 1565 Drive working with 1581 or even CMD-FD2000 Disks. This would make my archiving easier quite much. For now the internal disk drive cant be used at all from the core.

sy2002, 2023-10-14:
Thank you for your feature request. Indeed we plan to make use of the built-in "1565" in future .Our minimum goal is to use it as an 1581; CMD-FD2000 would be the cherry on the cake.

(Workaround: In the meantime, as this might take a while: You are aware that you can connect real IEC devices, for example a real 1581 or CMD-FD2000 to the MEGA65? Just make sure that the device ID is not 8.)

oe3gwu, 2023-10-14:
Thank you @sy2002, I am aware of this, and I use my CMD-2000 Clone (Firedrive 2k) with the Emulator. Works nice.