MJoergen/C64MEGA65 #94: Consider supporting "Snappy Rom" Cartridge which does not work with C64 core

paich64, 2023-10-14:
Snappy rom is an extended Super Snapshot V5 which comes either as a rom or as a cartridge: 

https://github.com/adrianglz64/snappyrom

It comes with all the possible required tools to code like in the good old days.

The binaries rom/.crt are available in the attached release.

[snappyrom534.zip](https://github.com/MJoergen/C64MEGA65/files/12904563/snappyrom534.zip)

sy2002, 2023-10-15:
Thank you @paich64 - sounds interesting! Yes, we should support this. Did you test it in the `*.crt` simulation mode - or - for example on an EasyFlash3 (real hardware)?

paich64, 2023-10-16:
Hi @sy2002 I do not have easyflash 3, so i only tested the *.crt simulation mode.

sy2002, 2024-10-30:
@MJoergen If this is "not dangerous" as defined for 5.2, then this might be an interesting one to do either for you or for me.

MJoergen, 2026-04-17:
Fixed in V6 Alpha Release 4