MJoergen/C64MEGA65 #96: RESEARCH/DECIDE: Audio glitch

sy2002, 2023-10-20:
From a Skype conversation between @MJoergen and @sy2002:

**MJoergen:**

At some point we might want to look into the C64 core's SID implementation. While debugging my audio controller I noticed some strange effects from some of the games I played. Specifically, what should be a smoothly varying signal from the SID had some mysterious discontinuities, i.e. large jumps from one sample to the next. I have a slight suspicion that there is something wrong with the audio filters.

I therefore added the option to generate a simple sine wave (test tone) of a fixed frequency, and that works like a charm. I even fed the analog output into my desktop machines microphone input and fired up the program audacity and did a spectral analysis! The waveform looks smooth as a baby's bum :-)
 So I'm confident my audio controller works as expected.
 
 **sy2002:**
 
 AARGH: A SID problem?  Hopefully it is "just" the filter: Can you do this test again and switch-off "Audio-improvements"? This "Audio-improvements" is some strange MiSTer "thing" that we never looked-at more closely.
 
 **Next steps:**
 
1. First of all, @MJoergen should re-do the very test with the test signal but with DEACTIVATED MiSTer filters, so that we can see: Is this a SID bug or a MiSTer audio filter bug? 

2. If it is a SID issue:
* We need to rename this issue and then probably move to a future release (post 5.1). Sorgelig of MiSTer completely re-did large part of the SID in late 2022, see this list: https://github.com/MiSTer-devel/C64_MiSTer/commits/master  There are five SID related commits between November 16 and November 25. Back in the days, we (AmokPhaze, deft and sy2002) tested using audio recordings of MiSTer and our current SID version for various songs and both SID variants (6581 and 8580) if and what sorgelig's updates did to the sound: WE WERE NOT ABLE TO HEAR ANY DIFFERENCES between "before and after".
* But maybe the refactoring sorgelig did, which includes using code from the this project: https://github.com/daglem/reDIP-SID, nevertheless fixes the issue that @MJoergen detected.
* If it is NOT a SID issue, then let us NOT touch the SID implementation we currently use.

3. If it is a MiSTer audio filter issue:
* Let us find out what this audio filter actually does
* Let us decide if we can/want to fix it or if the filter needs to go away (or at least is DISABLED by default)

MJoergen, 2023-10-20:
[image]

The above screenshot is a dump of the digital output from the SID (i.e. the signal `c64_sid_l` in main.vhd). The sample rate is approx 1 MHz, so the screenshot covers approx 16 ms of real time. In this test I have disabled the "Audio improvements" feature in the On Screen Menu.

So the image shows many discontinuous jumps, i.e. almost vertical lines.

sy2002, 2023-10-21:
> It's possible that the output shown is simply the addition of a "nice" sine wave and a square wave.

The SID output sounds rather normal to my ears. Can you please look at this YouTube playlist: https://youtube.com/playlist?list=PLXhLeiiveJmNhFf5ShVwwXspGfgt-ww8c&si=HjcqibEL-4RIaTTt

It contains oscilloscope views of various SID songs, but not the summarized wave, but the three voices of the SID separately. To my layman's eye it looks like, if I "added these waves in my head to one summary wave form": Wouldn't this look quite like what you found?

If you feel the same, then we might "Close as not planned" this issue.

Or we can have people do some more sophisticated comparison tests between V5's SID output and V5.1's output.

MJoergen, 2023-10-21:
I agree. I was concerned that I might have made a mistake with the signed/unsigned confusion, but that seems not to be the case.
The youtube video shows that my understanding of the SID output is very much lacking.
In other words, I think we're good.
=> Closing.

MJoergen, 2023-10-21:
.

closed 2023-10-21 by MJoergen