MJoergen/C64MEGA65 #97: R3: Remove R4 signals

sy2002, 2023-10-21:
If this is intentional for some reason, then please use the "Close as not planned" button.

The `top_mega65-r3.vhd` file contains various signals, that are not available on an R3 board. Examples:

```vhdl
   -- Audio DAC. U37 = SSM2518CPZ-R7
   audio_mclk_o       : out   std_logic;
   audio_bick_o       : out   std_logic;
   audio_sdti_o       : out   std_logic;
   audio_lrclk_o      : out   std_logic;
   audio_pdn_n_o      : out   std_logic;
```

I think there are even more for examples of R4 signals in the R3 file: The I2C bus and the PMOD header, but honestly I am not sure if these are available on R3 boards or not. Please double-check.

Also please double-check if more files are affected such as XDC or others.

I also opened an issue in the M2M framework so that in case that this will be fixed, that it will be fixed on both sides of the equation.

MJoergen, 2023-10-21:
This is indeed intentional. The R3 board DOES have an audio DAC. The output from that is connected to the two (optional) internal speakers.
Interestingly, the DAC on the R4 board is changed compared to the R3 board.

MJoergen, 2023-10-21:
I've gone over the R3 schematic and added all signals / pins that are connected to the FPGA. Many of these are not used, but I decided to add them to the XDC file and the top-level file for consistency.

closed 2023-10-21 by MJoergen