MJoergen/C64MEGA65 #98: Make use of bi-directional joystick ports (R4/R5)

sy2002, 2023-10-21:
Newer MEGA65 boards support bi-directional joystick ports.

At this moment, we are not yet aware how to make use of this feature and which hardware uses it so that we have decent test cases.

MJoergen, 2023-12-19:
Do we have some way of testing this? It's straight-forward to implement.

sy2002, 2024-01-03:
"Upcoming protopad": https://www.protovision.games/shop/protopad/protopad.php?language=de

dillof, 2024-08-26:
The easiest way is probably to connect an LED with resistor from one of the pins to ground, then you should be able to toggle it on/off by setting the bit high/low.

Here is a list of hardware that I'm aware of that uses bi-directional joystick ports, unfortunately none of them are easy to purchase:

- Trap Them Controller (sold out)
- Protovision Protopad (still in prototype stage)
- real time clock on CMD SmartMouse / SmartTrack (rare)
- Ninja SNES Pad (extremely rare)
- Multijoy (PCB available on [their site](http://www.multijoy.net))
- Atari CX21/CX50 ([reproduction PCB](https://github.com/tebl/A2600-Keyboard) available)

All of these (except the SmartMouse) are supported by my controller test program [Joyride](https://github.com/T-Pau/Joyride). 

If you have a SNES controller and a SNES controller connector, it's easy to build an adapter to make it behave like the Trap Them controller. If you want, I can build one and send it to you. I have the CX21 PCBs, I can build one of those for you as well.

Alternatively, if you want me to test a pre-release core, I'd be happy to. I have an R6 mainboard and most of the hardware above.

sy2002, 2024-08-28:
Thank you, @dillof - this info is highly appreciated. It will take us a while until we tackle this one, but I am optimistic that we will tackle it eventually.

sy2002, 2024-10-30:
More info here: https://kugelblitz360.github.io/C64MEGA65DOCS/joysticks-paddles-mice.html#bi-directional-devices

@MJoergen possible not hard to do and therefore maybe a candidate for 5.3 but @Kugelblitz360 is there enough value "in the market" for that?  Gentlemen, please share you thoughts here. Currently still tagging it as 6.0 or later.

dillof, 2024-10-30:
Since this is one of the major enhancements of the R6 board, it would be nice to also support it in the C64 core.

sy2002, 2024-10-30:
@dillof Thank you for chiming in. Can you recommend a device for me to buy, a source where to get it and some software that makes use of it, so that I (or @MJoergen) can test it? I need to admit, that when I read Boris' analysis in the new documentation (https://kugelblitz360.github.io/C64MEGA65DOCS/joysticks-paddles-mice.html#bi-directional-devices) I was a bit turned-off by the obscurity. But still: Willing to move on here - maybe you can supply a few weblinks here.

Kugelblitz360, 2024-10-30:
I do not own ANY device that would use the functionality so it is impossible for me to test and really hard to judge. Looking at the install base of maybe 1000 C64 Cores currently, I would guess that we'd only find a handful of people that actually own these. Protovision must have abandoned the Protopad - would that ever go into production, we might have a good reason.

dillof, 2024-11-10:
All controllers are supported by my program Joyride. As for regular software: There is a GEOS driver for the SmartMouse real time clock. The Trap Them Controller was developed for the game of the same name. Multijoy and CX21 are Atari devices that happen to also work on the C64, I'm not aware of software for it on the C64.

sy2002, 2026-02-16:
**Seems that we already started to implement this feature during V5.1 but did not finish**

I stumbled over this commit from @MJoergen from October 2023, which is part of our core since Version 5.1: https://github.com/MJoergen/C64MEGA65/commit/6316782d459ca8afc654cf7f284c5d255d06e44e  "R4: Add support for bi-directional joystick"

I do not fully understand what this does, but I do assume it is correct to route the joystick input signals to the CIAs, which it what we are doing here. At least it did NOT generate problems since 5.1:

[image]

What we are still have as "TBD" is the output to make it bi-directional, so the commit above seems to be half-way to the goal:

[image]

And obviously, we also need to check if these output signals are routed to the bi-directional joystick port of R4, R5, R6 machines.