MJoergen/C64MEGA65 #101: Test after dynamic clock and other refactorings

sy2002, 2023-11-02:
Hi Michael, on Skype you wrote this here:

===
* MFJ/MH: Verify/Test latest development on our R3 and R4 boards. (One thing to test is the REU+HDMI, e.g. the TreuLove demo. Another thing to test is the reset button, and the HDMI resolution switching, in conjunction with various demos)
===

Here is how we should proceed:

1. Please test on your R4 including REU+HDMI

2. THEN assign this to me. I will:

a) Test on my R3

b) If (1) and (2a) positive: Ask @paich64 to test using his R3A and his much better eye for detail: Our refactoring might have broken timings that you can only see, if you run TreuLove in parallel to a C64 and/or to a MiSTer. @paich64 can do that.

sy2002, 2023-11-02:
@paich64 Can you please test, if TreuLove runs as good as you are used to run it on V5 using this Alpha 3 version of the V5.1 version of the core? [c64v51a3.zip](https://github.com/MJoergen/C64MEGA65/files/13244494/c64v51a3.zip) I do see some white horizontal lines at the very beginning but I think I remember this was always the case. The goal of this test is: Not worse than V5.

paich64, 2023-11-02:
@sy2002 just flashed core v5.1 A3 and ran TreuLove on both V5.1A3 and V5.0final : The runs are strictly identical :)
horizontal lines at the very beginning (during text scrolling) have always been there and from what i has grabbed from the net, they are also visible when runnning the demo on a true C64 with a truc REU1750. So it's all ok :)

sy2002, 2023-11-03:
THANK YOU @paich64  - this is awesome :-)

closed 2023-11-03 by sy2002