MJoergen/C64MEGA65 #104: Implement support for bi-directional cartridge reset (R5 specific)

sy2002, 2023-11-02:
Implement support for bi-directional cartridge reset (R5 specific) - expected outcomes:

* We won't need the heuristics for the EF3 any more and the EF3 might be even more compatible
* We won't need the "reset workaround" for the Kung Fu Flash (KFF) any more

sy2002, 2024-01-02:
@paich64 Please perform the following tests with this Alpha 5 version of the core on both: R3 and R5. But if R3 is not possible for some reason - then R5 is much more important: [C64MEGA65-V5.1A5.zip](https://github.com/MJoergen/C64MEGA65/files/13814650/C64MEGA65-V5.1A5.zip)

1. Start a game on EF3. Expected behavior: Works on R3 and R5
2. Press reset while the game is running (long reset might be necessary). Expected behavior: Brings you back to EF3's main menu on both, R3 and R5
3. Start a freezer cartridge on EF3. Expected behavior: Works on R3 and R5
4. Start an alternate KERNAL on EF3. Expected behavior: Works only on R5 and does not work on R3
5. Use the KFF to start an EasyFlash cartridge image (.crt). Expected behavior: R3 needs a workaround as described in doc/cartridges.md while R5 should work like a charme
6. Flash something to the EF3 and then start it. Expected behavior: Works on both, R3 and R5

paich64, 2024-01-03:
**EF3 tests (on R5) :**
1. Start a game on EF3. Expected behavior: Works on R3 and R5                 =>         **OK on R5**
2. Press reset while the game is running (long reset might be necessary). Expected behavior: Brings you back to EF3's main menu on both, R3 and R5                                                                              =>         **OK on R5** 
3. tart a freezer cartridge on EF3. Expected behavior: Works on R3 and R5  =>         **OK on R5** 
4. Start an alternate KERNAL on EF3. Expected behavior: Works only on R5 and does not work on R3 => **KO on R5**

I can flash a custom kernal successfully, but when selecting it, the "standard" c64 kernel will be loaded instead.
I have tested the cartridge on my true C64 and the custom kernal will load successfully

6. Flash something to the EF3 and then start it. Expected behavior: Works on both, R3 and R5 => **OK on R5**

closed 2024-01-03 by sy2002