MJoergen/C64MEGA65 #105: Remove 60Hz option for PAL

sy2002, 2023-11-02:
Why remove the 60Hz option for PAL? Quote from a Skype conversation from me to MJoergen:

"I want to REMOVE the 60Hz screen mode that we already offer, from the C64 core. The 60Hz mode flickers as hell as we only support PAL for the time being. The reason why I offered a 60Hz mode for the C64 was that I hoped it would increase compatibility for some of the poor souls who did not see anything on their screens. Time showed us: It was not 50Hz was 60Hz - it was the "+5V issue" that made the cut. So the 60Hz abilities that you were adding are meant for the GameBoy, for Arcade cores and - in the C64 context - of course for the NTSC mode that we have on our ROADMAP.md (maybe for V6?)."

Long story short: 60Hz modes are for NTSC on the C64, therefore we will remove them for the time being as our current core is PAL-only.

MJoergen, 2023-11-02:
Since you mention the breaking of backwards compatibility regarding the config files, then that might suggest we postpone this for V6. I like the idea that V5.1 is essentially identical to V5 without any major changes (except "trivial" bugfixes).

sy2002, 2023-11-02:
OK - agreed. Moving this to V6.