MJoergen/C64MEGA65 #110: Internal floppy drive keeps spinning when floppy is inserted

MauriceMEGA65, 2023-11-13:
In C64 core (tested with V5 and 5.1 alpha), the floppy drive keeps spinning when a floppy is inserted.
Info here on Discord: https://discord.com/channels/719326990221574164/794775503818588200/1173362893924864061

The floppy drive behaves normally in MEGA65 core.

MJoergen, 2023-11-14:
Hmm. In V5 the FDC pins are unconnected, and likewise in V5.1A1. However, since V5.1A2 the FDC pins are all tied to ground. So @MauriceMEGA65 , do you know **which** V5.1 alpha you see this on? It says so in the OSM, on the "about" page at the bottom.

MauriceMEGA65, 2023-11-14:
Not at my M65 at the moment, but I believe it was alpha 2. But the same issue I see on the regular V5

MauriceMEGA65, 2023-11-14:
Ok, there are grounding differences? OK, let me check this morning

MauriceMEGA65, 2023-11-14:
Version 5.1A3, see also the movie in the link above

MJoergen, 2023-11-14:
I've confirmed the same behavior on my R4 board.

closed 2023-11-14 by MJoergen

MauriceMEGA65, 2023-11-15:
Good to know you can reproduce the issue! And already solved?
If you have a new alpha version to test then please let me know if you want me to test it.

sy2002, 2023-11-15:
@MauriceMEGA65 : Yes @MJoergen was able to reproduce it and solved it. If you want to test: @MJoergen Can you post a bitstream here? I can turn it into a .cor for @MauriceMEGA65 .

sy2002, 2023-11-19:
@MauriceMEGA65 Can you confirm that this core fixes the spinning floppy for you? 
[c64m65v51-d4d294b.zip](https://github.com/MJoergen/C64MEGA65/files/13404569/c64m65v51-d4d294b.zip)

MauriceMEGA65, 2023-11-20:
Yes, solved!!

But something else I just discovered: Pac-man needed to have the hard reset to be fixed, which is done in one of the alpha versions.
However, I just encountered the following:
1. Hard-reset DOES properly reset the system, I'm back in the BASIC 2 or JiffyDOS screen
2. When I do a soft-reset AFTER the hard-reset, I'm back in the Pac-man welcome screen!?!?

sy2002, 2023-11-20:
@MauriceMEGA65 Thank you for sharing your feedback about this issue here: Internal floppy drive keeps spinning when floppy is inserted. We cannot mix different topics in one issue. This goes contrary to the concept of an issue tracker, so we will not follow up on your Pacman observations here.