MJoergen/C64MEGA65 #115: Implement ethernet carts in the emulation

M3wP, 2023-11-17:
Please consider adding devices such as the RR-NET and 64NIC+ to your emulation.  

Here is the link for the RR-NET:  https://icomp.de/shop-icomp/en/produkt-details/product/rr-net-mk3.html

Thanks!

sy2002, 2023-11-17:
@M3wP Thank you for the suggestion. Interesting. I personally have no idea what you can do on a C64 with Ethernet (or Internet?) => Can you elaborate some uses cases and also software and online sites that support browsing C64's ?

M3wP, 2023-11-17:
A big one is the OS Contiki https://www.c64-wiki.com/wiki/Contiki 

The game 8-bit Slicks https://8bit-slicks.com/

I also have Yahtzee! of my own creation https://github.com/M3wP/M3wPYahtzee

I think it is untapped potential.

lgblgblgb, 2023-11-17:
I believe there was some kind of BBS software as well, which can "dial in" to a modern BBS (for real a TCP/IP accessible stuff over the Internet), cannot remember the name though. But there is even "geolink" stuff like iRC client running in GEOS using the IP65 TCP/IP stack using RR-NET. Which is kinda cool, though I've never seen that in real life, to be honest ...

sy2002, 2023-11-18:
@M3wP and @lgblgblgb Thank you, gentlemen. This is great context.

lgblgblgb, 2023-11-21:
By the way, a more complete list (scroll down a bit for the list):

https://www.protovision.games/hardw/rr_net.php?language=en

Some other info:

https://www.c64-wiki.com/wiki/RR-Net

SvOlli, 2023-11-22:
I've written an alternative firmware for the RR-Net MK3. The original intend was to use it to upload code during development. But that's expanded a bit, like it can download code using PXE.

So I can help on testing, as long as we've got support for Action Replay cartridges.

sy2002, 2023-11-22:
Thank you for the additional insights @lgblgblgb and @SvOlli . This project here is more for the long haul but we will gladly ping you when time is ripe.

sy2002, 2026-08-20:
@xlar54 - any updates on 64NIC+ ?

M3wP, 2026-08-20:
I have a game client that runs on the C64 and I have now ported it to the M65 if this helps any.

xlar54, 2026-08-21:
Sorry, got tied up.  Will test this tonight and update!

xlar54, 2026-08-21:
Insta-update.. it detects the cart, gets DHCP, resolves DNS, even connects, but having an issue actually transmitting data.   i double checked that the network connection was good.  using Guruterm as baseline.  Standby for more info.

xlar54, 2026-08-21:
Disregard.  I was sharing a wifi connection from another machine.  Direct connect to the router, and connections work perfectly.