MJoergen/C64MEGA65 #117: Mouse button not working in EoB on R5

ki-bo, 2023-11-30:
Just tested with V5.1A4 on R5. Can load Eye of the Beholder crt just fine, but when using a 1351 mouse in port 2 the mouse button does not react. I can still move the mouse, though. This 1351 works fine on R3a with V5.

ki-bo, 2023-11-30:
Interesting, the mouse button not working is only happening on port 2 with EoB. If I connect the 1351 to port 1 and use "Flip joystick ports" everything is working fine. EoB only seems to support mouse in port 2, though, so need to use that flip option.

ki-bo, 2023-11-30:
The port issues seem to be more general but depending on whether a crt is loaded or the REU is active. I used this test program:

`10 PRINTPEEK(56320):GOTO10`

I can see it reacting fine on all joystick directions and the fire button in port 2 if Expansion Port setting is "Use Hardware Slot". As soon as I switch it to "Simulate 1750 REU 512KB", the test program stops working (not reacting at all to joystick port 2 input). Switching back to "Use hardware slot" (even without reset) makes it work again.

sy2002, 2023-12-01:
Comment from Paul on Skype:

>Also note that the joystick ports might mis-behave if the bi-di control of those ports has not yet been implemented in a core, or at least holding the drive-joy-lines-low signals are not held at the correct state to tri-state those drivers.

paich64, 2023-12-02:
same issue for me with port 2 with v5.1A4 on R5. Same workaround (plugging joystick in port 1 and switching port). Issue does not occure with V5.1A4 on R3A.

MJoergen, 2023-12-03:
Same comment above applies to the signal `i_framework/joy_2_fire_n_i`.

MJoergen, 2023-12-03:
Mysteriously, the same applies to the port `fb_fire_n_i_IBUF`. No changes are detected on the corresponding output port `fb_fire_n_o_OBUF`; it remains '1' in all cases.
Likewise the signal `joystick_5v_disable_o_OBUF` is always '0' and the signal `paddle_drain_o_OBUF` is always '1'.

closed 2023-12-15 by MJoergen