MJoergen/C64MEGA65 #118: POWER CARTRIDGE does not reset on R5 V5.1 A4

deftmega, 2023-12-12:
POWER CARTRIDGE only works on first/cold start but does not show logo nor any functionality after pressing reset on R5 with core version V5.1 A4. It has been verified to fully work on R4 with the same core.

Side note: Power Cartridge III seems to also behave randomly on reset an I expect that to be the same issue.

sy2002, 2023-12-13:
@deftmega Thanks for reporting the Alpha 4 issue. Alpha 5 is supposed to fix cartridge bugs - can you retest and report back here in GitHub?

sy2002, 2024-01-02:
@MJoergen Can you please test if this Alpha 5 is fixing the issue: [C64MEGA65-V5.1A5.zip](https://github.com/MJoergen/C64MEGA65/files/13814650/C64MEGA65-V5.1A5.zip)

MJoergen, 2024-01-03:
I've confirmed that KCS Power Cartridge partially works with the supplied V5.1A5 on an R5 board. I've tested with the core written to flash.

- :white_check_mark:  After power on, KCS cartridge logo is shown and cartridge works as expected.
- :white_check_mark:  Pressing reset on cartridge brings up its reset menu. It works as expected.
- :question: Pressing SHORT reset on MEGA65 resets C64, but no KCS logo is shown. The BASIC extensions (e.g. the `MONITOR` command) still works. I'm not sure, whether the KCS logo is supposed to be shown or not, i.e. what the expected behavior is.
- :x: Pressing LONG reset on MEGA65 resets C64, but no KCS logo is shown. However, the BASIC extensions (e.g. the `MONITOR` command) no longer works. The cartridge appears disabled. The cartridge reset button does still function, but I then have to choose the "RESET" option in the cartridge menu in order to re-enable the `MONITOR` command.

sy2002, 2024-01-03:
It's been a while - I am not sure what the right semantics are. But I do know that the V5 core on R3/R3A shows the correct behavior.

MJoergen, 2024-01-06:
Here are my observations on the R3 board using V5.0:

1. Power on : KCS logo is visible. `MONITOR` command works.
2. I enter a short infinite loop BASIC program and run it.
3. SHORT reset on MEGA65 : Machine is reset, BASIC program is gone, KCS logo is visible, `MONITOR` command works.
4. I enter a short infinite loop BASIC program and run it.
5. LONG reset on MEGA65 : Same as SHORT reset.
6. I enter a short infinite loop BASIC program and run it.
7. Cartridge reset on KCS Power Cartridge brings up the cartridge menu:
    - `CONTINUE` : BASIC program resumes.
    - `BASIC` : Return to BASIC prompt, but BASIC program is still there, and can be restarted.
    - `RESET` : Machine is reset, BASIC program is gone, KCS logo not visible, `MONITOR` command works.

MJoergen, 2024-01-06:
Here are my observations on the R3 board using V5.1:

1. Power on : KCS logo is visible. `MONITOR` command works.
2. I enter a short infinite loop BASIC program and run it.
3. SHORT reset on MEGA65 : Machine is reset, BASIC program is gone, KCS logo is visible, `MONITOR` command works.
4. I enter a short infinite loop BASIC program and run it.
5. LONG reset on MEGA65 : **Machine is crashed**. Recover by power cycling.
6. I enter a short infinite loop BASIC program and run it.
7. Cartridge reset on KCS Power Cartridge brings up the cartridge menu:
    - `CONTINUE` : BASIC program resumes.
    - `BASIC` : Return to BASIC prompt, but BASIC program is still there, and can be restarted.
    - `RESET` : Machine is reset, BASIC program is gone, KCS logo not visible, `MONITOR` command works.

So, basically same as V5.0, except for the highlight in step 5.

sy2002, 2024-01-06:
@MJoergen For me this means "works as intended" and we can close this issue. What are your thoughts?

MJoergen, 2024-01-06:
Still need to test on R5. The above two results were to establish the "base line".

sy2002, 2024-01-06:
Ahh - OK - I see. Thank you.

MJoergen, 2024-01-14:
Here are my observations on the R5 board using V5.1:

1. Power on : KCS logo is visible. `MONITOR` command works.
2. I enter a short infinite loop BASIC program and run it.
3. SHORT reset on MEGA65 : Machine is reset, BASIC program is gone, **KCS logo is NOT visible, `MONITOR` command DOES NOT work**.
4. Power cycle. I enter a short infinite loop BASIC program and run it.
5. LONG reset on MEGA65 : Machine is reset, BASIC program is gone, **KCS logo is NOT visible, `MONITOR` command DOES NOT work**.
6. Power cycle. I enter a short infinite loop BASIC program and run it.
7. Cartridge reset on KCS Power Cartridge brings up the cartridge menu:
    - `CONTINUE` : BASIC program resumes.
    - `BASIC` : Return to BASIC prompt, but BASIC program is still there, and can be restarted.
    - `RESET` : Machine is reset, BASIC program is gone, KCS logo not visible, `MONITOR` command works.

So, basically same as R3, except for the highlight in steps 3 and 5.

sy2002, 2024-01-15:
Your results sound like "sending" the reset signal from the core to the cartridge does not work on R5, right? This might be a bug in my code or a different timing behavior on R5 (maybe due to R5's logic the signal is "shorter" or "later"?) or a bug in the hardware. We better found this out before the R6 production run starts...

MJoergen, 2024-01-15:
The following waveform shows the application of SHORT reset on the R5 board.

Notice how `cart_reset_i` goes from asserted (0) to de-asserted (1). This shows the state of the physical RESET pin on the cartridge port, specifically the net `F_C64_RESET` in the schematic, i.e. just after the level-converter U60. This suggests that the reset signal is correctly propagated to the cartridge.

Next we see that EXROM is low, and that ROML is asserted multiple times. This shows that the CPU is reading the `CBM80` signature, and then executing code from `$8000`. It's not yet clear, whether this data is coming from the cartridge or from RAM. But a `PRINT PEEK(32768)` at the BASIC prompt gives the value 0, which is not what the CPU was reading (it was reading the value `$09`).

[image]

The next waveform shows the situation slightly EARLIER, i.e. just when the cartridge reset goes from de-asserted (1) to asserted (0).

Note how when BASIC is running (and the KCS logo is visible), that EXROM is low, and ROML is continuously asserted, i.e. the CPU is executing code directly from the cartridge at `$8000`. There is also an almost imperceptible "glitch" in the GAME signal, just after `cart_reset_i` is asserted.

[image]

MJoergen, 2024-01-17:
Here is a waveform obtained by:

* Holding MEGA65 reset button pressed
* Programming the FPGA via JTAG
* Set trigger to rising edge of cart_reset_i
* Releasing MEGA65 reset button

[image]

We see that GAME and EXROM are both asserted at this time.

MJoergen, 2024-01-17:
The cartridge has a flip-flop (U11, 74LS175) that drives the GAME and EXROM signals (via pins 7 and 15). The RESET pin on the cartridge is connected directly to the reset pin of this flip-flop, see the schematic below:

[image]

Therefore, it should be the case that while RESET is held low, then GAME and EXROM should both be low as well. However, this is not the case, as seen in the above ILA traces.

gardners, 2024-01-18:
Do you have an oscilloscope, with which you can probe the /RESET pin on the cartridge port to see if it goes low when you tell the FPGA to pull it low? I'd consider making a super-simple bitstream that only does that, to confirm that basic function, and see what voltage /RESET goes to when you command it low.

MJoergen, 2024-01-18:
I made another waveform using the ILA. This time, I'm connecting the ILA probe pins directly to the I/O pins of the FPGA. That's why you see the IBUF and OBUF, which Vivado has appended automatically.

[image]

The result is the same. RESET output is asserted, but GAME input is not.

I will next try to measure the voltage directly on the cartridge port.

sy2002, 2024-01-19:
@MJoergen wrote to @gardners 

When reset is de-asserted, the voltage is 5.1 V. This is ok. However, when RESET is asserted by the FPGA, i.e. when both F_C64_RESET and F_C64_RESET_EN are pulled low, then the voltage on the cartridge reset pin (C64_RESET) is 1.584 V. This is way outside spec. According to the data sheet for the chip 74LS175 (on the KCS Power Cartridge), the "Low Level Input Voltage" must be below 0.8 V. So no wonder it's not working.

The question now is, why am I measuring such a high voltage on the reset pin ?

Note: I'm measuring the RESET voltage on the input pin of the 74LS175 on the KCS cartridge, because that is more convenient. I've visually checked the PCB traces on the cartridge, that this pin is directly connected to the RESET pin on the cartridge port.

So what next? Would it make sense for you to do a similar measurement on your board? I mean, am I just unlucky? I think not, because Deft could not get the Power cartridge to work on his R5 board either.

According to the R5 schematic, the C64_RESET net is pulled high to 5V by a 4.7k resistor, and driven low by a SN74AHCT1G125 buffer. This buffer has a "Low-level output current" of 8 mA, which should be more than enough.

sy2002, 2024-01-19:
And here is additional info from @MJoergen :

> Without the cartridge I read 0.05 V on the RESET pin. With the cartridge inserted (and doing the exact same test), I read 1.5 V on the RESET pin.

My conclusion:

* The R3A board does not have this issue

* The R5/R6 board's mechanism that pulls the RESET line (and I assume this is true also for NMI and IRQ, others?) to GND  is not "strong enough" when it comes to pulling to GND while the R3A board - or a normal Commodore 64 - absolutely is.

We need to check this with Trenz.

MJoergen, 2024-01-20:
I did another set of measurements (on R5 board with cartridge inserted), measuring between U59 pin 4 and the cartridge RESET pin.

* Switch off, the resistance  = 22.3 ohm
* Switch on, assert RESET. The voltage difference = 0.789 V

That means that through this resistor runs a current of 0.789 / 22.3 = 35 mA.

Additionally, I measured resistance between RESET and VCC on the cartridge port.

* Without cartridge = 4.703 kΩ
* With cartridge = 4.604 kΩ

This can be interpreted as the cartridge acts like an equivalent shunt resistance of approx 220 kΩ.

sy2002, 2024-01-22:
I am copy/pasting stuff here that @MJoergen posted on Skype which seems additive to what is written above in https://github.com/MJoergen/C64MEGA65/issues/118#issuecomment-1902247615 just to make sure we are not losing information:

Regarding the RESET line and the KCS cartridge: I've measured directly on U59 the following (relative to pin3 = GND):
pin 1 (/OE) = 0.0 V
pin 2 (input) = 0.0 V
pin 4 (output) = 0.78 V
pin 5 (VCC) = 5.1 V

Now, according to the specs for this chip, "Low level output voltage" should be less than 0.44 V, and even that is only with a load current of 8 mA.

I am not able to measure directly on the resistor R271, but apparently it has a voltage drop of (1.6 V - 0.78 V) = 0.82 V over it. That is a current of 0.82 V / 20 ohm = 40 mA.

Now, it is possible that I've messed up the measurements, but the bottom line is that the cartridge is not working, and that it is related to the RESET line.

========================

Another measurement **I also repeated the measurement on R4 board, and with the cartridge inserted I measured the RESET voltage level to 0.8 V, which is ok. And different from R5.**

==========================

Idea: Are we sure they used the correct resistor value ? I mean, if instead of 20 ohm they someone used, say, 2k resistor. Then that would in fact fit nicely with the observations: A voltage divider between 4.7k and 2k would indeed give approx 1.6 V over the smaller resistor.

Scratch that last hypothesis. I used the multimeter (with the R5 board switched off) and measured the resistance between the RESET line and U59 pin 4. I got the value 22.3 ohm.

======
NEXT POST
======

I just redid the measurements (on R5 board with cartridge inserted), measuring between U59 pin 4 and the cartridge RESET pin.

* Switch off, the resistance  = 22.3 ohm
* Switch on, assert RESET. The voltage difference = 0.789 V

I.e. same results as before, but this time the voltage difference was measured directly.

So, I would really like to know why there is a current of 35 mA through this resistor. ...

sy2002, 2024-01-22:
[image]

sy2002, 2024-01-22:
[image]

I had a closer look at my KCS. It seems that the schematic is not describing the circuitry on my cartridge. The following are some (bad) photos of the top and bottom. There is quite some annoying glare on the photos, but we can make out a few key points:

On the photo of the bottom side, the lower right corner shows the underside of a 16-pin chip. We clearly see on the photo that pin 1 of the chip is connected to the RESET pin of the cartridge (as well as to a 14-pin chip further up). And we also see that pins 6 and 10 of this 16-pin chip are connected together. I've tested with the multimeter that these two pins are indeed connected.

This is bad news, because on the 74LS175 pins 6 and 10 are both outputs.

First conclusion: In the schematics I found AND in the schematics Paul has found, the 74LS175 does not have pins 6 and 10 connected. So my board is connected differently.

Second conclusion: Since these are output pins on the 74LS175, it's quite likely not an 74LS175 at all. Unfortunately, the part numbers of the various chips are unreadable. Yes, I looked under the small capacitor on the top side - nothing written there.

Third conclusion(?): The RESET pin is as mentioned also connected to a 14-pin chip at the top. This is not present in the schematics either.

The top side of the board clearly shows the text "1985/86". I wonder if the schematics are for a later version of the cartridge. I note that my board matches closely the photo shown at the top of this page : https://rr.pokefinder.org/wiki/Power_Cartridge

sy2002, 2024-01-22:
[image]

sy2002, 2024-01-22:
[image]

sy2002, 2024-01-22:
The status of the workaround is the following:
* With the KCS Power cartridge I can perform a short RESET and the cartridge restarts
  correctly. This works > 30 times in a row.
* With the KCS Power cartride, a long RESET also restarts the cartridge. This is
  unexpected, because the "CBM80" signature should be masked away.
* With the EF1CR cartridge, short reset is now flaky, only working sometimes.
* With the EF1CR cartridge, long reset doesn't work.  Core doesn't start up at all.

So, more work is still to be done. 

The workaround is here:

https://github.com/MJoergen/C64MEGA65/commit/221a5aacc41631e55acf26d66c46c2f5869e893c

gardners, 2024-01-23:
It may well make sense to get some of these made up, so that we can more effectively test problems like this:

https://github.com/svenpetersen1965/C64---Cart64out

I may order some PCBs for this, and look at building them up.

gardners, 2024-01-23:
I have ordered 10 of those PCBs. I should have them by around 30 Jan 2024, so will be better able to investigate things after that, if I can get hold of a KCS cartridge here in AU. Olivier is going to send the KCS he has ordered to me when it arrives to him, quite likely arriving early Feb.

sy2002, 2024-01-23:
Here are additional comments/findings from @MJoergen to Trenz that @MJoergen has sent via E-Mail:

On the R4 board (where the KCS cartridge is working as expected) the RESET pin on the cartridge port is driven by U9 (SN74LVCH16T245DG). According to the datasheet, this buffer has a maximum value for "Continuous output current" of +/- 50 mA. Furthermore, the recommended value for "Low-level output current" is 32 mA. It seems the cartridge is drawing (or rather pushing) a current of 35 mA into the RESET pin. Since this is almost within spec, this might explain why the cartridge works on the R4.

However, on the R5 board, the RESET pin is driven by U59 (74AHCT1G125). According to the datasheet there, the maximum value for "Continuous output current" is +/- 25 mA, and a recommended value for "Low-level output current" of 8 mA. So the performance of the output buffer on R5 is significantly lower (worse) than for the R4 board.

MJoergen, 2024-01-25:
Trenz reported back that they had measured the output characteristics of the chip 74AHCT1G125 using the following circuit:
[image]
The result is as follows:
[image]
The conclusion is that the internal resistance of the buffer is approximately 25 Ohms, in the range 0...40 mA.

MJoergen, 2024-01-25:
The fix agreed upon is to replace the seven instances of the 22 ohm resistor with a 0 ohm resistor. This does not require new PCB.

MJoergen, 2024-01-25:
I measured using a thermal camera the temperature of the U59 chip (when the 22 ohm resistor is shorted). Even with continuous assertion of the RESET pin, I only saw an increase in 1 - 2 degrees celsius. Much less than e.g. the FPGA itself.

MJoergen, 2024-01-25:
With the 22 ohm resistor shorted and asserting RESET (with the KCS Power Cartridge inserted), I measured a voltage of 1.07 V on the RESET pin. This is still slightly out of spec for TTL chips (should be less than 0.8 V), but it works better than before, and it's the best we can do.

It seems the KCS Power cartridge is acting like a 100 ohm impedance connected to +5V.

MJoergen, 2024-01-25:
Later revisions of the MEGA65 might change the output driver to one with a lower internal resistance (less than 20 ohm). That will drive the RESET line to less than 0.8 V, even with the KCS Power Cartridge.

sy2002, 2024-01-26:
We can close this issue: The production R6 boards will have R269, R270, R271, R272, R273, R274, R275 replaced by 0 Ohm bridges. This will resolve the issue. I will document in doc/models.md that unmodded R5 boards are having this problem and point to this issue for reference.

closed 2024-01-26 by sy2002