MJoergen/C64MEGA65 #119: Better HDMI Flicker-Fix

sy2002, 2023-12-13:
Taken from 2ab483fdccdc63ecf9b407cdc3e9ab513dce7cbc:

```vhdl
-- An even better alternative is to cascade two MMCM's: To get from 100 MHz to 31.4496 MHz
-- we need a factor of 0.31449600. Written as a fraction that is 4914/15625 = 2*3^3*7*13 / 5^6.
-- We can factor this fraction into two factors: 2*3*13/5^3 = 0.624 and 3^2*7/5^3 = 0.504.
-- These two factors can we implemented as follows:
-- 0.504 : CLKFBOUT_MULT_F = 47.25, DIVCLK_DIVIDE = 5, CLKOUT0_DIVIDE_F = 18.75
-- 0.624 : CLKFBOUT_MULT_F = 19.50, DIVCLK_DIVIDE = 1, CLKOUT0_DIVIDE_F = 31.25
-- Note: It's necessary that the first MMCM is 0.504, to keep f_VCO of the second MMCM
-- within the range of 600 - 1200 MHz.
-- With the above approach we get the exact clock frequency required, and therefore no
-- longer need any dynamic shifting of phase.
```

sy2002, 2026-02-19:
@MJoergen Are you sure? Looks like it was fixed on one of your research branches and never made it to the final product? [mfj_issue_119](https://github.com/MJoergen/C64MEGA65/compare/mfj_issue_119)  I lost track if this one made it to a release. And if it is actually an improvement worth merging. Can you please double-check?  (Re-opening for that reason.)

MJoergen, 2026-04-25:
This has now been merged into the develop branch. No Alpha Release yet.

This fix is purely an internal optimization and should give no visible change for the users.

closed 2026-05-03 by sy2002