MJoergen/C64MEGA65 #120: Advanced C64 Compatibility Settings

sy2002, 2024-01-02:
* Create a "Advanced Compatibilty Settings" submenu
* Move "CIA: Use 8521 (C64C)" to it
* Add settings for the VIC-II as tagged with `@TODO sy2002` in `fpga_sid_iec.vhd`, particularly NMOS, HMOS, old HMOS
* Sift through MiSTer's advanced config settings and offer settings if applicable
* Find out if MiST is doing something with the `mode6572` and `mode6567old` setting of the VIC-II
* SID?
* CIA?
* PLA?
* <other advanced settings?>

MJoergen, 2024-11-11:
Perhaps the DVI feature (#121) belongs here too ...

sy2002, 2024-11-12:
> Perhaps the DVI feature (#121) belongs here too ...

Well @MJoergen, this is less a C64 specific compatibility setting such as "which NMOS/CMOS type of SID do we want" or "which CIA" - but more a general HDMI setting; I guess the DVI topic is more a MiSTer2MEGA65 topic from which the C64 core then benefits.