MJoergen/C64MEGA65 #121: DVI: Autodetect monitor capabilities and auto-activate DVI mode

sy2002, 2024-01-07:
If one has a display / monitor that does not support HDMI but only DVI, then the screen stays black when the core is started. Our current "workaround" for this situation is this recipe that users can run through:

https://github.com/MJoergen/C64MEGA65#hdmi-compatibility

Using M2M's new I2C features, we can find out the monitor's capabilities and automatically activate DVI mode for such displays.

MJoergen, 2024-11-10:
Looking at the wikipedia page https://en.wikipedia.org/wiki/Extended_Display_Identification_Data it seems that byte 20, bits 3-0 can be used to differentiate between a HDMI monitor and a DVI-only monitor.

MJoergen, 2024-11-10:
@sy2002 : In order to test this, it would be nice to have the model number of a specific monitor that will benefit from this fix.

MJoergen, 2024-11-10:
The current version of M2M (in V5.1) supports QNICE access to the HDMI I2C EDID rom, as follows:

1. Write 0x0005 to 0xFFF4 (access I2C controller)
2. Write 0x0350 to 0xFFF5 (access I2C address 0x50 on HDMI)
3. Read from 0x7114 (direct I2C memory map access to register 0x14)

This is exactly the byte number 20 mentioned in the previous comment.

MJoergen, 2024-11-10:
Data point for my monitor "Philips 272S1" (which does support sound):

The DisplayPort connector on the monitor returns the value 0xa5. Bits 3-0 => DisplayPort
The HDMI connector on the (same) monitor returns the value 0x80. Bits 3-0 => Undefined

sy2002, 2024-11-10:
> @sy2002 : In order to test this, it would be nice to have the model number of a specific monitor that will benefit from this fix.

@MJoergen I asked the community for support. See this Discord message: https://discord.com/channels/719326990221574164/801767398675316756/1305176385836417055

Piramania, 2024-11-10:
Samsung 2433SW (DVI+VGA) here :)

Schwefelholz, 2024-11-10:
Lenovo LT1913pA
Wincor-Nixdorf PA10

Both DVI and VGA

sy2002, 2024-11-12:
Thank you @tonedef71 for this info via Discord:

I have a Sceptre 19" VGA monitor that supports DVI.

sy2002, 2024-11-12:
Thank you to Matt84 for this info via Discord:

I have a benq bl3200pt that has dual link dvi input that works at 1440p I could test with

Kiwi-Phoenix, 2024-11-15:
Dell P2714H (DVI & VGA) - This one seems to work fine.
Apple Cinema HD Display (DVI) - Does not work with Mega65 but is fine with Windows & Macs.
Both using HDMI-DVI cable.

jwa1974, 2024-12-20:
Have a Philips 242B1G (DVI / HDMI / VGA) if that would be of any help.